platformio/platformio-vscode-ide #9: Don't replace default terminal with PlatformIO

GeorgeSapkin, 2017-05-31:
It extremely annoying to have PlatformIO's `export PATH=...` and `pio --help` pop up each time a new terminal is opened.

ivankravets, 2017-05-31:
Please don't click on our Terminal button in toolbar. Use Ctrl + `

GeorgeSapkin, 2017-05-31:
That's what happens in a freshly opened project when I press Ctrl+`. I should have made it clear in original comment.

ivankravets, 2017-05-31:
What to do if a user opens a new terminal and tries to use `pio` command and it doesn't work?

closed 2017-06-03 by ivankravets

GeorgeSapkin, 2017-06-04:
Has the update been published yet?

ivankravets, 2017-06-04:
https://github.com/platformio/platformio-vscode-ide/releases/tag/v0.3.0

anweiss, 2017-06-12:
Just started seeing this issue again after upgrading to the May release of VSCode.

VSCode: 1.13.0
PIO VSCode: 0.3.0

commit da8b6fa3 (2017-06-03): Don't replace default terminal with PlatformIO // Resolve #9