platformio/platformio-vscode-ide #10: PlatformIO Terminal throws errors when opening

techMod, 2017-06-01:
I just installed the IDE for vscode. When I want to open the terminal it throws the following erros:

````
+ ... werShell\v1.0\;C:\Program Files\Git\cmd;C:\Program Files (x86)\NVIDIA ...
+                                                               ~~~
    + CategoryInfo          : ObjectNotFound: (x86:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

C:\Users\Stauffer\AppData\Local\Programs\Python\Python36-32\Scripts\ : Die Benennung "C:\Users\Stauffer\AppData\Local\Programs\Python\Python36-32\Scripts\" wurde nicht als Name eines Cmdlet, einer Funktion, einer Skriptdatei oder eines
ausführbaren Programms erkannt. Überprüfen Sie die Schreibweise des Namens, oder ob der Pfad korrekt ist (sofern enthalten), und wiederholen Sie den Vorgang.                                                                               I
n Zeile:1 Zeichen:864                                                                                                                                                                                                                       +
 ... hysX\Common;C:\Users\Stauffer\AppData\Local\Programs\Python\Python36- ...                                                                                                                                                              +
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   + CategoryInfo          : ObjectNotFound: (C:\Users\Stauff...n36-32\Scripts\:String) [], CommandNotFoundException
   + FullyQualifiedErrorId : CommandNotFoundException
                                                                                                                                                                                                                                            C
:\Users\Stauffer\AppData\Local\Programs\Python\Python36-32\ : Die Benennung "C:\Users\Stauffer\AppData\Local\Programs\Python\Python36-32\" wurde nicht als Name eines Cmdlet, einer Funktion, einer Skriptdatei oder eines ausführbaren     P
rogramms erkannt. Überprüfen Sie die Schreibweise des Namens, oder ob der Pfad korrekt ist (sofern enthalten), und wiederholen Sie den Vorgang.                                                                                            In
 Zeile:1 Zeichen:933                                                                                                                                                                                                                       +
... 32\Scripts\;C:\Users\Stauffer\AppData\Local\Programs\Python\Python36- ...                                                                                                                                                              +
                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  + CategoryInfo          : ObjectNotFound: (C:\Users\Stauff...on\Python36-32\:String) [], CommandNotFoundException
  + FullyQualifiedErrorId : CommandNotFoundException
                                                                                                                                                                                                                                           C:
\Users\Stauffer\AppData\Local\Programs\Python\Launcher\ : Die Benennung "C:\Users\Stauffer\AppData\Local\Programs\Python\Launcher\" wurde nicht als Name eines Cmdlet, einer Funktion, einer Skriptdatei oder eines ausführbaren           Pr
ogramms erkannt. Überprüfen Sie die Schreibweise des Namens, oder ob der Pfad korrekt ist (sofern enthalten), und wiederholen Sie den Vorgang.                                                                                            In
Zeile:1 Zeichen:994                                                                                                                                                                                                                       + .
.. ython36-32\;C:\Users\Stauffer\AppData\Local\Programs\Python\Launcher\ ...                                                                                                                                                              +
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 + CategoryInfo          : ObjectNotFound: (C:\Users\Stauff...ython\Launcher\:String) [], CommandNotFoundException
 + FullyQualifiedErrorId : CommandNotFoundException
                                                                                                                                                                                                                                          C:\
Users\Stauffer\AppData\Local\Microsoft\WindowsApps : Die Benennung "C:\Users\Stauffer\AppData\Local\Microsoft\WindowsApps" wurde nicht als Name eines Cmdlet, einer Funktion, einer Skriptdatei oder eines ausführbaren Programms         erk
annt. Überprüfen Sie die Schreibweise des Namens, oder ob der Pfad korrekt ist (sofern enthalten), und wiederholen Sie den Vorgang.                                                                                                      In Z
eile:1 Zeichen:1052                                                                                                                                                                                                                      + ..
. n\Launcher\;C:\Users\Stauffer\AppData\Local\Microsoft\WindowsApps;C:\ ...                                                                                                                                                              +
              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
+ CategoryInfo          : ObjectNotFound: (C:\Users\Stauff...oft\WindowsApps:String) [], CommandNotFoundException
+ FullyQualifiedErrorId : CommandNotFoundException
                                                                                                                                                                                                                                         C:\P
rogram : Die Benennung "C:\Program" wurde nicht als Name eines Cmdlet, einer Funktion, einer Skriptdatei oder eines ausführbaren Programms erkannt. Überprüfen Sie die Schreibweise des Namens, oder ob der Pfad korrekt ist (sofern    entha
lten), und wiederholen Sie den Vorgang.                                                                                                                                                                                                 In Ze
ile:1 Zeichen:1106                                                                                                                                                                                                                      + ...
 Users\Stauffer\AppData\Local\Microsoft\WindowsApps;C:\Program Files\L ...                                                                                                                                                              +
                                                    ~~~~~~~~~~                                                                                                                                                                              +
 CategoryInfo          : ObjectNotFound: (C:\Program:String) [], CommandNotFoundException                                                                                                                                                   +
 FullyQualifiedErrorId : CommandNotFoundException
                                                                                                                                                                                                                                        x86 :
 Die Benennung "x86" wurde nicht als Name eines Cmdlet, einer Funktion, einer Skriptdatei oder eines ausführbaren Programms erkannt. Überprüfen Sie die Schreibweise des Namens, oder ob der Pfad korrekt ist (sofern enthalten), und  wieder
holen Sie den Vorgang.                                                                                                                                                                                                                 In Zeile:1 Zeichen:1150                                                                                                                                                                                          [... cut ...] ound: (pio:String) [], CommandNotFoundException                                                                                                                                                          + FullyQualifiedErrorId : CommandNotFoundException
                                                                                                                                                                                                                                      PS X:\sensor>

```

marlboro, 2017-06-01:
I have the same issue and it has something to do with the PATH.  PlatformIO works fine (compiles and uploads), but if you try to view the terminal it is flooded with errors.  The first line is trying to set PATH. 
 The last one I can see (it won't let me scroll) is shown below.

```
PS D:\dev\source\Lab\mcu\stm\hacky> pio --help
pio : The term 'pio' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was included, verify that the path is correct and try
again.
At line:1 char:1
+ pio --help
+ ~~~
    + CategoryInfo          : ObjectNotFound: (pio:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
```

techMod, 2017-06-02:
I added the following entries to PATH:

C:\Program Files (x86)\Microsoft VS Code\bin **(should be a 64 bit version, no?)**
C:\Users\me\.platformio\penv\Scripts
C:\Program Files\LLVM\bin

Now the serial monitor and the terminal work, but they throw all the errors first. Probably PlatformIO is not trying to recognize the entries as a part of it?

techMod, 2017-06-02:
I added the following entries to PATH:

C:\Program Files (x86)\Microsoft VS Code\bin **(should be a 64 bit version, no?)**
C:\Users\me\.platformio\penv\Scripts
C:\Program Files\LLVM\bin

Now the serial monitor and the terminal work, but they throw all the errors first. Probably PlatformIO is trying to recognize the entries as a part of it?

Also `PlatformIO: Add Include Path to Settings` won´t work.

marlboro, 2017-06-02:
I wanted to add that I am running on Windows 10 x64.  I also have Platform IDE for Atom installed, which may be part of the problem.  Because the Atom version was installed first, it's first in my path.

The pertinent user path entries are:

```
C:\Users\mnorris\AppData\Local\atom\bin
C:\Program Files (x86)\Microsoft VS Code\bin
```

ivankravets, 2017-06-02:
We know about this issue, will fix as soon as possible. Sorry.

commit f1ffcb3d (2017-06-03): Fix issue with PIO Terminal and Windows PS // Resolve #10
closed 2017-06-03 by ivankravets