platformio/platformio-vscode-ide #13: 0.2.0->0.3.0 stuck in upgrade loop

hoppke, 2017-06-05:
Hi, I have pio core 3.3.1 installed (standalone).
After upgrading to pio-vscode-ide 0.3.0 vscode gets stuck in a loop where, after opening vscode, it first displays "verifying platformio installation" in the status bar, then goes on to 
"Installing PlatformIO Core... Please don't close this window and don't open other folders until this process is completed." in the terminal.

After that it asks me to restart vscode, and then the whole process repeats - verifying, installation, restart.
0.2.0 worked fine.

ivankravets, 2017-06-05:
Could you try to remove standalone PIO Core? Open system `cmd`

```
pip uninstall platformio
```

`platformio-ide.useBuiltinPIOCore = true`, `platformio-ide.customPATH = null`

hoppke, 2017-06-05:
Removing the standalone version made it work again. The environment switched from the PIO in my Python directory to the PIO in my $HOME/.platformio/penv

Interestingly, my standalone PIO was 3.3.1 (and on "pio update/upgrade" it would claim that 3.3.1 is the latest version). But the version that got installed in my $HOME is 3.4.0b10

Also, before removing my standalone PIO I tried to launch the vscode extension under debugger, and it immediately threw
Error: Incompatible PIO Core 3.3.1
    at c:\Users\grzegorz\Desktop\platformio-vscode-ide-develop\lib\installer\stages\platformio-core.js:313:15
(...)

So I guess that explains it. I suppose 3.3.1 is the latest stable, but the vscode plugin needs a newer build?
(anyway, I'm happy to leave my env the way it is - it works now, issue solved, thx!)

ivankravets, 2017-06-05:
I've just tried to reproduce this issue and can't :( 

Could you try again to install 3.3.1, then start VSCode. When it asks about to reload, please open PIO IDE Terminal (icon on toolbar) and copy paste here `echo %PATH%`.

Thanks!

hoppke, 2017-06-06:
OK, so my system-wide PATH included both Python/Scripts and ~/.platformio/penv/Scripts.
The Python directory was listed first.

If I remove ~/.platformio from my system PATH, leaving just the Python path, then within vscode I get ~/.platformio/... prepended to PATH and everything works as expected, it ends up running the bundled PIO. I suppose the extension modifies $PATH to facilitate the running of PIO commands from the vscode terminal?

If I keep ~/.platformio/... on my system PATH though, the extension will not prepend it, so it always ended up after my Python PIO path, and never got invoked.

I'm and edge case :)

ivankravets, 2017-06-06:
Thank you very much!!! That is why I like the community and open source!

commit 8f6fd698 (2017-06-06): Force built-in PIO Core in environment PATH // Resolve #13
closed 2017-06-06 by ivankravets