platformio/platformio-vscode-ide #14: Changing the baud rate for the serial monitor

JohnDoeKyrgyz, 2017-06-06:
How do I change the baud rate for the serial monitor? A similar plugin fixed this by adding a setting that you can set for each project.

[https://github.com/formulahendry/vscode-platformio/issues/3](https://github.com/formulahendry/vscode-platformio/issues/3)

ivankravets, 2017-06-06:
The correct solution is https://github.com/platformio/platformio-core/issues/787

We will implement it within the next few days

ivankravets, 2017-06-06:
Resolved in https://github.com/platformio/platformio-core/commit/b04fc327c06e13ed7cafac5135269990326cac4e

Please `pio upgrade`

ivankravets, 2017-06-06:
http://docs.platformio.org/en/latest/projectconf.html#monitor-options

closed 2017-06-06 by ivankravets

sleepdefic1t, 2017-10-21:
Yeah. that still isn't helping.

Can you maybe... Idk...
Explain the random shit you linked to or what the issue is?
None of that was even helpful.

I want to like PIO.. **Want to**..
But it's slow as hell and never works without a crapload of tweaking.

Back to Arduino IDE, I suppose..

sleepdefic1t, 2017-10-21:
yeah.
Just tried for another 20 minutes, then uninstalled it.
It's crazy that people actually pay for this.

danielbibit, 2018-01-10:
Erh... as explicit was in the link provided by @ivankravets adding 'monitor_baud = 115200' to the .ini file works perfectly fine. Pretty handy as I use non-standart baud rates on some applications. Thank you @ivankravets !

VioletGiraffe, 2018-02-17:
Why does it have to be hard-coded in the platform settings, rather than selectable in the UI as is the case with Arduino IDE and Visual Micro (Visual Studio add-in for embedded development)?

zouden, 2018-02-24:
Yeah I agree, this issue is not resolved until there's a UI for selecting the baud rate like there is in the Atom extension. This is the second time I've had to use google to find out how to change the baud rate in VSCode. It needs to be discoverable.

ArtiomLK, 2018-10-26:
In the file _**platformio.ini**_ use:
`monitor_speed` instead of `monitor_baud`

> Warning! `monitor_baud` option is deprecated and will be removed in the next release! Please use `monitor_speed` instead.

For example:
`monitor_speed = 115200`

zouden, 2018-10-27:
Coming back to this problem (making the baud setting discoverable), an easy solution would be to make all new projects explicitely have `monitor_speed = 9600` in the platformio.ini file by default. Then it's obvious how to change it. @ivankravets what do you think?

ivankravets, 2018-10-27:
@zouden we've already started to do that. See https://github.com/platformio/platform-espressif32/blob/develop/examples/espidf-hello-world/platformio.ini

We can't add `monitor_speed` to each project because we don't know about default baudrate used in a framework. What could really resolve this issue, this is GUI for `platformio.ini`. We plan to start a work on it soon. You will be able to manage all `platformio.ini` options directly from PIO Home.

ee-muhammad, 2019-02-04:
@ivankravets After all my efforts on google, I came across only four parameters related to Serial Monitor which are configurable through platformio.ini file (https://docs.platformio.org/en/latest/projectconf/section_env_monitor.html). What about configuring other parameters like eol through platformio.ini?

ivankravets, 2019-02-04:
You can create a custom task. See https://docs.platformio.org/en/latest/ide/vscode.html#custom-tasks

exocode, 2019-02-12:
It's not working on Mac Newest VSCode newest platformio plugin.

```
 platformio upgrade                                                                                 ✔ 
You're up-to-date!
PlatformIO 4.0.0a3 is currently the newest version available.
```

**platformio.ini**

```
[env:nanoatmega328]
monitor_speed = 115200
platform = atmelavr
board = nanoatmega328
framework = arduino
```

ivankravets, 2019-02-13:
How did you check it? Do you use the same baudrate in code and `PlatformIO.ini`?

exocode, 2019-02-13:
I see it in the Terminal `platformio device monitor --baud 9600`

exocode, 2019-02-13:
Tschuldigung, ich nochmal:

Kann ich - sollte ich mal "genug gespart" haben, diese Linsen (EV-200D) dann auch für die Fatsharks verwenden? Oder brauche ich dann wieder andere?

Danke

Jan

ivankravets, 2019-02-13:
It seems that you hard coded this command somewhere. We don’t  use extra option for monitor command

exocode, 2019-02-13:
I found it: It is was "hard coded" in the settings of the Visual Code Editor for platformio plugin:
I changed it here and it worked (y) @ivankravets thank you for the hint!

You can see everything here:

[image]

ivankravets, 2019-02-18:
@exocode You have duplicated PlatformIO extensions. Please remove https://github.com/formulahendry/vscode-iot-utility extension and use only `PlatformIO IDE`.