platformio/platformio-vscode-ide #15: Setting PATH for terminal fails

AndreKR, 2017-06-07:
My original problem was that the `pio` command couldn't be found when I clicked the "Platform IO: Serial Monitor" or "Platform IO: New Terminal" buttons.

I investigated a bit, but my findings confuse me.

So I replaced [this block](https://github.com/platformio/platformio-vscode-ide/blob/develop/src/terminal.js#L23-L30) with:
```
    console.log(process.env.PATH);
    console.log(constants.IS_WINDOWS);
    
    if (constants.IS_WINDOWS) {
      commands.push('set PATH=' + process.env.PATH);
    } else if (process.env.SHELL && process.env.SHELL.includes('fish')) {
      commands.push('set -gx PATH ' + process.env.PATH.replace(/\:/g, ' '));
    } else {
      commands.push('export PATH=' + process.env.PATH);
    }
    
    commands.push('blah');
    commands.push('set FOO=BAR');
    commands.push('set PATH=' + process.env.PATH);
    commands.push('echo %FOO%');
    commands.push('set FOO=BAR');
    commands.push('echo %FOO%');
    commands.push('echo %PATH%');
    commands.push('pio --help');
```

Here's the console output:

> ```
> [Extension Host]
> C:\Users\andre\.platformio\penv\Scripts;C:\Windows\SYSTEM32;C:\Windows\SYSTEM32;C:\Windows;C:\Windows\SYSTEM32
> extensionHost.ts:285
> [Extension Host]
> true
> extensionHost.ts:285
> ```

And here's the terminal output:

> ```
> Microsoft Windows [Version 6.1.7601]
> Copyright (c) 2009 Microsoft Corporation. Alle Rechte vorbehalten.
> 
> C:\dev-projects\#Arduino\vscodetest>echo %FOO%
> %FOO%
> 
> C:\dev-projects\#Arduino\vscodetest>set FOO=BAR
> 
> C:\dev-projects\#Arduino\vscodetest>echo %FOO%
> BAR
> 
> C:\dev-projects\#Arduino\vscodetest>echo %PATH%
> C:\Windows\SYSTEM32;C:\Windows
> 
> C:\dev-projects\#Arduino\vscodetest>pio --help
> Der Befehl "pio" ist entweder falsch geschrieben oder
> konnte nicht gefunden werden.
> 
> C:\dev-projects\#Arduino\vscodetest>
> C:\dev-projects\#Arduino\vscodetest>
> 
> ```

ivankravets, 2017-06-07:
Could you try replace 
```
commands.push('set PATH=' + process.env.PATH);
```
with
```
commands.push(`set PATH="${process.env.PATH}"`);
```
?

AndreKR, 2017-06-07:
Same result.

ivankravets, 2017-06-07:
Could you try to replace `PATH` with `Path` ?

AndreKR, 2017-06-07:
Wait a second... I found that I had some spaces in my PATH and it seems to depend on the length of PATH, checking...

AndreKR, 2017-06-07:
Argh, I am unable to reproduce it anymore. What I did:
I set my system PATH to different things. At some point I copied my PATH hardcoded to the `command.push`. I removed some characters, suddenly it worked. Then I added the characters again, it still worked. Then I added more characters, it stopped working again.
I was playing around to find a pattern and now it is always working.
It seems to be possible to break by putting in a very long path (more than 1100 characters). I didn't find an exact limit though, due to the kind of "hysteresis" behavior described above.
I'm sorry I lost the evidence, but for me it works right now.

ivankravets, 2017-06-07:
:)

https://support.microsoft.com/en-us/help/830473/command-prompt-cmd.-exe-command-line-string-limitation

ivankravets, 2017-06-07:
I'll think of a workaround. Thanks a lot for your research! 👍

commit f07c044f (2017-06-28): Added workaround for Windows command-line string limitation // Resolve #15
closed 2017-06-28 by ivankravets