platformio/platformio-vscode-ide #25: Enable local echo in PySerial

rmcolbert, 2017-07-14:
CTRL+E is a default mapping in VS Code. To enable local echo, this keybinding needs to be added to the PlatformIO extension:
    { "key": "ctrl+e", "command": "", "when": "terminalFocus" }

This lets the CTRL+E pass thru to the terminal.

ivankravets, 2018-05-30:
Sorry for the delay. It works for me now with the latest VSCode. Please re-open if it still does not work for you.

closed 2018-05-30 by ivankravets

jasonblewis, 2018-12-05:
on linux, platformio seems intercept Ctrl+E when the terminal has focus. I'm using core v 3.6.2

rlogiacco, 2018-12-21:
It doesn't work for me, CTRL+E doesn't pass through to the terminal when serial terminal has focus.
It would be nice to have a dedicated setting: probably most of us want it enabled by default.

ivankravets, 2018-12-26:
You can control the monitor from a command line. See https://docs.platformio.org/en/latest/userguide/cmd_device.html#platformio-device-monitor

rlogiacco, 2018-12-26:
I’m not sure I understand correctly: are you suggesting I should stop using the plug icon to start the serial monitor and rather start launching it from command line providing all the options I like?

ivankravets, 2018-12-26:
You can actually create a custom task and assign hot keys. See https://docs.platformio.org/en/latest/ide/vscode.html#task-runner

rlogiacco, 2018-12-26:
That is great, thank you!
I believe though, the option will be very helpful nonetheless, if that’s not too complex to add.

ivankravets, 2018-12-26:
Will be it useful if we implement advanced serial monitor with logging, etc in PIO Home? Or CLI version is better?

rlogiacco, 2018-12-26:
I’m not sure I understand the question... I believe local echo should be enabled by default as it helps a lot: if that’s not viable, I believe it should be very easy to turn it on... That’s my point of view, if it makes sense