platformio/platformio-vscode-ide #26: Taskrunner Warning at v1.14

sstaub, 2017-07-15:
When building, a warning popup comes up "No task is currently running"
[image]

ivankravets, 2017-07-15:
Hi @dbaeumer,

This issue is caused by [these lines](https://github.com/Microsoft/vscode/blob/969ffcb995c98a512baa5cc257ba444650fc69ae/src/vs/workbench/parts/tasks/electron-browser/task.contribution.ts#L1970). Do we have API for `getActiveTasks()`? Currently, we run `workbench.action.tasks.terminate` before each PlatformIO command because we don't know which task was run last. For example, if it was Serial Monitor, we should terminate it becasue other tasks, such as: upload & monitor will not work.

--------

Please note that I've moved to Dynamic Tasks today. See 
- https://github.com/platformio/platformio-vscode-ide/blob/develop/src/tasks.js

Can this help us to fix this issue?

P.S: Having `Pre/Post` callbacks in `Task Interface` would be amazing!

Thanks in advance!

sstaub, 2017-07-16:
Works fine, thank you

commit 5178d915 (2017-07-17): Remove "No task is currently running" warning // Resolve #26
closed 2017-07-17 by ivankravets