platformio/platformio-vscode-ide #27: PIO and VSCode; spaces in paths in tasks.json

transentia, 2017-07-15:
This follows on from https://community.platformio.org/t/pio-and-vscode-spaces-in-paths-in-tasks-json/2257/3

Tasks.json appears to be an auto-generated file and contains:

```
{
    "version": "0.1.0",
    "runner": "terminal",
    "command": "\"C:\\Users\\Bob Brown\\.platformio\\penv\\Scripts\\platformio.exe\"",
    "isShellCommand": false,
```

Note that the 'command' value is a path containing a space.

This causes a build to fail:

```
C:\Users\Bob : The term 'C:\Users\Bob' is not recognized as the name of a cmdlet, function, script file, or operable
program. Check the spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ C:\Users\Bob Brown\.platformio\penv\Scripts\platformio.exe -c vscode  ...
+ ~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (C:\Users\Bob:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
```

Since tasks.json is regenerated frequently, any attempt to edit it does not 'stick'.

transentia, 2017-07-15:
I note that launch.json may have a similar issue, since it contains:

`"gdbpath": "C:/Users/Bob Brown/.platformio/penv/Scripts/piodebuggdb",`

and also gets overwritten regularly.

ivankravets, 2017-07-16:
Could you try this? 

https://github.com/platformio/platformio-vscode-ide/issues/26#issuecomment-315562497

transentia, 2017-07-16:
I replied at https://community.platformio.org/t/pio-and-vscode-spaces-in-paths-in-tasks-json/2257

I tried 0.5.0-beta.1; didn't solve the problem.

Apologies. I think half of my comments are ending up on github and half on platformio. Not sure what I'm doing wrong! I THOUGH i'd replied on community.platformio.org...

ivankravets, 2017-07-16:
1. Please open PlatformIO IDE Terminal (see last  icon on bottom toolbar) and run `pip install -U https://github.com/platformio/platformio-core/archive/feature/no-vscode-tasks.zip`
2. Remove `settings.json` which you mentioned on forum
3. Restart VSCode. Now, it should not create `tasks.json` in `.vscode` folder.
4. Menu > Tasks > Build

Does it work now?

transentia, 2017-07-17:
Ran the pip command you gave
Manually removed settings.json
Manually removed tasks.json
Restarted vscode
After about 10 secs, a new tasks.json is created.

The attached image shows result and relevant versions in use.

[image]

transentia, 2017-07-17:
Thought I'd try creating a NEW project from scratch, just in case...no difference.

For completeness:

```
PS C:\Users\Bob Brown\Desktop\XXX> pio --version
PlatformIO, version 3.4.1a5
PS C:\Users\Bob Brown\Desktop\XXX>
PS C:\Users\Bob Brown\Desktop\XXX> pio init --board uno --ide vscode

The current working directory C:\Users\Bob Brown\Desktop\XXX will be used for project.
You can specify another project directory via
`platformio init -d %PATH_TO_THE_PROJECT_DIR%` command.

The next files/directories have been created in C:\Users\Bob Brown\Desktop\XXX
platformio.ini - Project Configuration File
src - Put your source files here
lib - Put here project specific (private) libraries

Project has been successfully initialized!
Useful commands:
`platformio run` - process/build project from the current directory
`platformio run --target upload` or `platformio run -t upload` - upload firmware to embedded board
`platformio run --target clean` - clean project (remove compiled files)
`platformio run --help` - additional information
PS C:\Users\Bob Brown\Desktop\XXX> ls .\.vscode\

    Directory: C:\Users\Bob Brown\Desktop\XXX\.vscode

Mode                LastWriteTime         Length Name
----                -------------         ------ ----
-a----       17/07/2017  10:21 AM           8328 c_cpp_properties.json
-a----       17/07/2017  10:21 AM            426 launch.json

PS C:\Users\Bob Brown\Desktop\XXX> ls

    Directory: C:\Users\Bob Brown\Desktop\XXX

Mode                LastWriteTime         Length Name
----                -------------         ------ ----
d-----       17/07/2017  10:21 AM                .pioenvs
d-----       17/07/2017  10:21 AM                .vscode
d-----       17/07/2017  10:21 AM                lib
d-----       17/07/2017  10:21 AM                src
-a----       17/07/2017  10:21 AM             32 .gitignore
-a----       17/07/2017  10:21 AM           1566 .travis.yml
-a----       17/07/2017  10:21 AM            431 platformio.ini

PS C:\Users\Bob Brown\Desktop\XXX> "int main() { }" | ac src\main.cpp
PS C:\Users\Bob Brown\Desktop\XXX> pio run
[07/17/17 10:26:19] Processing uno (platform: atmelavr; board: uno; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 27 compatible libraries
Looking for dependencies...
Project does not have dependencies
Compiling .pioenvs\uno\src\main.o
Compiling .pioenvs\uno\FrameworkArduino\PluggableUSB.o
Compiling .pioenvs\uno\FrameworkArduino\Print.o
Compiling .pioenvs\uno\FrameworkArduino\Stream.o
Compiling .pioenvs\uno\FrameworkArduino\Tone.o
Compiling .pioenvs\uno\FrameworkArduino\USBCore.o
Compiling .pioenvs\uno\FrameworkArduino\WInterrupts.o
Compiling .pioenvs\uno\FrameworkArduino\WMath.o
Compiling .pioenvs\uno\FrameworkArduino\WString.o
Compiling .pioenvs\uno\FrameworkArduino\_wiring_pulse.o
Compiling .pioenvs\uno\FrameworkArduino\abi.o
Compiling .pioenvs\uno\FrameworkArduino\hooks.o
Compiling .pioenvs\uno\FrameworkArduino\main.o
Compiling .pioenvs\uno\FrameworkArduino\new.o
Compiling .pioenvs\uno\FrameworkArduino\wiring.o
Compiling .pioenvs\uno\FrameworkArduino\wiring_analog.o
Compiling .pioenvs\uno\FrameworkArduino\wiring_digital.o
Compiling .pioenvs\uno\FrameworkArduino\wiring_pulse.o
Compiling .pioenvs\uno\FrameworkArduino\wiring_shift.o
Archiving .pioenvs\uno\libFrameworkArduino.a
Linking .pioenvs\uno\firmware.elf
Building .pioenvs\uno\firmware.hex
Calculating size .pioenvs\uno\firmware.elf
AVR Memory Usage
----------------
Device: atmega328p

Program:     138 bytes (0.4% Full)
(.text + .data + .bootloader)

Data:          0 bytes (0.0% Full)
(.data + .bss + .noinit)

============================================= [SUCCESS] Took 1.84 seconds =============================================
PS C:\Users\Bob Brown\Desktop\XXX>
```

Note the version of PIO IS the latest you asked me to try.

You can see that the IDE created 'suboptimal' tasks.json, which--when run, as before--gives:

```
C:\Users\Bob : The term 'C:\Users\Bob' is not recognized as the name of a cmdlet, function, script file, or operable
program. Check the spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ C:\Users\Bob Brown\.platformio\penv\Scripts\platformio.exe -c vscode  ...
+ ~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (C:\Users\Bob:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
```
[image]

ivankravets, 2017-07-17:
1. You have multiple PIO Cores in a system. Please remove them. Open system `cmd` and run `pip uninstall platformio`
2. Please open PlatformIO IDE Terminal, BE CARAFEUL! I wrote that above! The last icon on toolbar (left bottom corner). You should see a title with PlatformIO in terminal.
3. Run here `pip install -U https://github.com/platformio/platformio-core/archive/feature/no-vscode-tasks.zip`

transentia, 2017-07-17:
That seems to have done the trick.

I THOUGHT I'd cleaned platformio from my system...i've certainly run the pip uninstall command before...a few times...still, maybe not.

MANY Thanks for your help on this. I'll close this now.

closed 2017-07-17 by transentia

ivankravets, 2017-07-17:
Thanks a lot! We will release official 0.5.0 today.