platformio/platformio-vscode-ide #28: IDE Doesn't work, vscode (1.14.2) PlatformIO IDE (0.5.1) on macosx (Sierra)

esherk, 2017-07-27:
In a new folder, I try to Initialize a new PlatformIO project and I get

```
Popup

PIP: Error: spawn /Volumes/Macintosh HD/Users/Eric/.platformio/penv/bin/pip ENOENT
```

and in output pane

```
Installing PlatformIO Core...
Please don't close this window and don't open other folders until this process is completed.
Failed to install PlatformIO IDE.
```

Files and directories are created in the folder, but the IDE is not present and none of the PlatformIO: commands work (can't open terminal, etc)

I have installed PlatformIO Core outside of PlatformIO IDE.  Before I did that, nothing was created in the directory.

I've also tried multiple uninstalls/reinstalls of all and can not get around this problem

ivankravets, 2017-07-27:
1. Open system terminal, not PIO IDE
2. `pip uninsntall platformio`
3. remove `/Volumes/Macintosh HD/Users/Eric/.platformio/penv`
4. Start VSCode

esherk, 2017-07-27:
Thanks, I had done that previously (except deleted all of .platformio).  

So, I did as you said (uninstall platformio and only remove penv dir) and on starting VSCode I get this in the output pane

`Installing PlatformIO Core...
Please don't close this window and don't open other folders until this process is completed.`

then the popup with

`PIP: Error: spawn /Volumes/Macintosh HD/Users/Eric/.platformio/penv/bin/pip ENOENT`

I press OK and in the output pane, this is added

`Failed to install PlatformIO IDE.`

is the (unescaped) space in my path causing an issue?

esherk, 2017-07-27:
That was the issue (space in PATH).  I renamed the disk (and also reduced PATH to /Users/Eric/...) and it worked.  I'll close this but maybe it needs to work its way into a fix (although it may be Python not the extension?) or into a FAQ/Troubleshooting section.

closed 2017-07-27 by esherk

fah, 2019-08-08:
I had the issue when I switched from Atom to Visual studio. 
Problem:
On opening any project - just a popup appeared, that the project is correctly setup and I have to use the cli to "platformio run". Button "OK". Nothing else. 
The solution was to "cd ~/ ; rm -r .platformio/penv;" and restart vs. That caused a reinstall of the platformio extension in VS. Now everything seems to work in VS.

ivankravets, 2019-08-11:
Please upgrade to the latest version of VSCoe and PlatformIO IDE extension 1.8.2 or above.