platformio/platformio-vscode-ide #29: PlatformIO IDE stop working

DaisukeUra, 2017-07-28:
Hello everyone.

I submit this problem in PIO forum but I thought that I should submit it here (because version number is < 1.0).

------

I use PlatformIO IDE on Visual studio code, OSX.

Build is possible immediately after starting the program.
But, when I modify some codes, save and build it, status bar displays **"Fetching build tasks..."** and it not complete never.
It recover only when reloading vscode after.

Is there a way to solve it?

Thanks.

ivankravets, 2017-07-28:
Please share your `platformio.ini` file here.

DaisukeUra, 2017-07-28:
My `platfomio.ini` is as follows.

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:genuino101]
platform=intel_arc32
board=genuino101
framework=arduino
```

Source file is in src directory.

DaisukeUra, 2017-07-28:
Disabled Arduino extension by Microsoft, it works correctly.
They seem to be processing something when saving.
I will use vscode while disabling the Arduino extension.

Thanks and sorry to disturb you.

closed 2017-07-28 by DaisukeUra