platformio/platformio-vscode-ide #30: travis.yml and gitignore files updated

dmaixner, 2017-08-02:
After saving platformio.ini my gitignore gets updated with those three lines:
`.pioenvs
.piolibdeps
.vscode`

Also file .travis.yml is created.

Why? I don't want this. How is this related to platformio as an iot platform? Why should platformio generate travis file? I would prefer to have my own control over gitignore (especially ".vscode" entry) and I don't want that travis file.

Both of these things should be either only in documentation or at least configurable.
Thank you.

ivankravets, 2017-08-02:
>  I would prefer to have my own control over gitignore (especially ".vscode" entry) 

Fixed in 
https://github.com/platformio/platformio-core/commit/578dfa40aa76c8eb8b35277f0277a74f039feecb
Please Tasks > Upgrade PIO Core

> Why? I don't want this. 

That is an old story. Will be resolved soon with https://github.com/platformio/platformio-core/issues/446

closed 2017-08-02 by ivankravets

derekmulcahy, 2018-09-17:
The second half of this issue, the creation of .travis.yml, still happens.

When a project is loaded init_ci_conf in init.py recreates .travis.yml. If I delete this file it is automatically recreated when I load the project in visual studio code.