platformio/platformio-vscode-ide #32: Do not destroy PIO Home when tab editor is changed

ivankravets, 2017-08-09:
Currently, VSCode totally destroys PIO Home when a user changes tab. Need to find a way how to restore previous IFRAME state.

There is no information here:

* https://code.visualstudio.com/docs/extensionAPI/vscode-api-commands#_working-with-the-html-preview

/cc @dbaeumer @tyriar should I open a feature request? Related [source code](https://github.com/platformio/platformio-vscode-ide/blob/develop/src/home.js#L18).

[image]

Tyriar, 2017-08-10:
/cc @mjbvz

ivankravets, 2017-08-17:
@mjbvz do you have any ideas?

mjbvz, 2017-08-17:
Yes, we destroy the html preview whenever the user switches away from it. This is by design.

The proper approach is write the html preview so that it can save its state and then recreate itself from this state. I believe the state needs to be saved by the extension code itself. Setting cookies in the webview doesn't work and I don't think local storage will either, but I'd have to check this.

For an example of this, take a look at the markdown extension: https://github.com/Microsoft/vscode/blob/master/extensions/markdown/media/main.js It doesn't really persist any state but it does set the initial view position using `vscode-markdown-preview-data` and it sends data back to vscode using `window.parent.postMessage`

ivankravets, 2017-08-17:
The problem here that our HTML Preview is a complex Web application which depends on the backend server. We need 1-2 secs to establish a connection.

2-nd problem, we don't render HTML content on-the-fly as Markdown does. Yes, in your case you can control these things. We have a pre-built Web application based on Rect components. That is a very expensive operation for us to "refresh" application.

Finally, if we have active session between Web-previewed APP and our backend (for example, physical USB Serial Monitor), we can't keep this session when a user switches between tabs.

In any case, thanks a lot for the info.

ivankravets, 2017-09-14:
Depends on https://github.com/Microsoft/vscode/issues/28263

ivankravets, 2018-05-05:
@mjbvz thank you so much for this API! 😊

commit e08f43bf (2018-05-05): Retain PIO Home state when switching between tabs // Resolve #32
closed 2018-05-05 by ivankravets