platformio/platformio-vscode-ide #34: Serial monitor UnicodeEncodeError: 'ascii' codec can't encode character u'\ufffd' in position 0: ordinal not in range(128)

chris-gunawardena, 2017-08-17:
When I open the Visual studio code serial monitor I get this error:
UnicodeEncodeError: 'ascii' codec can't encode character u'\ufffd' in position 0: ordinal not in range(128)

Works fine in the Arduino IDE Serial Monitor with the same baud rate.

MacOS with Python 2.7
$LANG en_GB.UTF-8

```
--- Miniterm on /dev/cu.usbserial-DN01J4DW  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Exception in thread rx:
Traceback (most recent call last):
  File "/usr/local/var/pyenv/versions/2.7/lib/python2.7/threading.py", line 530, in __bootstrap_inner
    self.run()
  File "/usr/local/var/pyenv/versions/2.7/lib/python2.7/threading.py", line 483, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/Users/chris/.platformio/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 453, in reader
    self.console.write(text)
  File "/Users/chris/.platformio/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 68, in write
    self.output.write(text)
UnicodeEncodeError: 'ascii' codec can't encode character u'\ufffd' in position 0: ordinal not in range(128)
```

kaline, 2018-03-02:
Me too
Traceback (most recent call last):
  File "/usr/lib64/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib64/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/lapada/.platformio/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 453, in reader
    self.console.write(text)
  File "/home/lapada/.platformio/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 68, in write
    self.output.write(text)
UnicodeEncodeError: 'ascii' codec can't encode character u'\u2402' in position 0: ordinal not in range(128)

ivankravets, 2018-10-10:
Could you try this? Does it work?
```
$ platformio device monitor -f printable
```

raffidil, 2018-11-18:
I have the same problem
`Exception in thread rx:
Traceback (most recent call last):
  File "/usr/lib64/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib64/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/raffi/.platformio/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 453, in reader
    self.console.write(text)
  File "/home/raffi/.platformio/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 68, in write
    self.output.write(text)
UnicodeEncodeError: 'ascii' codec can't encode character u'\u303d' in position 0: ordinal not in range(128)
`

chris-gunawardena, 2018-11-18:
This issue eventually went away, from memory I had to uninstall all other installs of platform-io. eg from cli using npm install etc. and only end up with the VSC one.

nickdex, 2019-01-05:
I've installed only the VScode version and getting this error. In normal operations serial monitor works, but this error comes up when
 - I reset my ESP8266
 - I do PlatformIO: Uload, (in cases when platformio does not automatically closes the monitor)
 - When I am running seven segment code, it won't run from the get go

```Console
--- Miniterm on /dev/ttyUSB1  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Exception in thread rx:
Traceback (most recent call last):
  File "/usr/lib64/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib64/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/nick/.platformio/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 453, in reader
    self.console.write(text)
  File "/home/nick/.platformio/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 68, in write
    self.output.write(text)
UnicodeEncodeError: 'ascii' codec can't encode character u'\u2400' in position 0: ordinal not in range(128)
```

koromkorom, 2019-01-23:
> Could you try this? Does it work?
> 
> ```
> $ platformio device monitor -f printable
> ```

```
$ platformio device monitor --baud 115200 -f printable
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun 8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
Exception in thread rx:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pi/.local/lib/python2.7/site-packages/serial/tools/miniterm.py", line 453, in reader
    self.console.write(text)
  File "/home/pi/.local/lib/python2.7/site-packages/serial/tools/miniterm.py", line 68, in write
    self.output.write(text)
UnicodeEncodeError: 'ascii' codec can't encode characters in position 0-4: ordinal not in range(128)
```

I have to say, that I used tmux and for this and then updated the esp32 in another thread. 
Still there should be a way to just ignore unknown characters ...

ivankravets, 2019-01-23:
Could someone provide a simple firmware to reproduce this issue?

koromkorom, 2019-01-24:
> Could someone provide a simple firmware to reproduce this issue?

Okay kind of false alarm, but still weird. The problem seems to be the pipe.
In my earlier post I took it out because I didn't think it mattered. 

platformio.ini
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
```

main.cpp
```
#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);
  Serial.print("Unknown utf8: ");
  Serial.println("\uFFFD");
}
```

```
platformio device monitor --baud 115200 | grep utf8
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Exception in thread rx:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pi/.local/lib/python2.7/site-packages/serial/tools/miniterm.py", line 453, in reader
    self.console.write(text)
  File "/home/pi/.local/lib/python2.7/site-packages/serial/tools/miniterm.py", line 68, in write
    self.output.write(text)
UnicodeEncodeError: 'ascii' codec can't encode character u'\ufffd' in position 13: ordinal not in range(128)
```

Ross-cz, 2019-04-10:
I had to patch write method to make utf-8 chars and invalid chars prevent this crash:
```
        try:
            self.output.write(text)
        except UnicodeEncodeError:
            self.output.write(text.encode("utf-8", "replace"))
```

rmgorman, 2019-04-30:
> Could someone provide a simple firmware to reproduce this issue?

I have the issue occur when attempting to run this example: https://github.com/platformio/platform-espressif32/tree/master/examples/espidf-coap-server

```
Exception in thread rx:
Traceback (most recent call last):
  File "/usr/lib64/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib64/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/ryan/.platformio/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 453, in reader
    self.console.write(text)
  File "/home/ryan/.platformio/penv/lib/python2.7/site-packages/serial/tools/miniterm.py", line 68, in write
    self.output.write(text)
UnicodeEncodeError: 'ascii' codec can't encode character u'\u241b' in position 0: ordinal not in range(128)
```

SvetoslavAsenov, 2019-05-25:
I have the same issue.
platformio.ini:
`[env:nanoatmega328]
platform = atmelavr
board = nanoatmega328
framework = arduino
lib_deps = Wire`

main.cpp:
`#include <Arduino.h>
#include <Wire.h>

void setup()

{
  Serial.begin (115200);
   while (!Serial)
    {
    }

  Serial.println ();
  Serial.println ("I2C scanner. Scanning ...");
  byte count = 0;
  pinMode(13,OUTPUT); 
  digitalWrite(13,HIGH);
  Wire.begin();
  for (byte i = 1; i < 120; i++)
  {
    Wire.beginTransmission (i);
    if (Wire.endTransmission () == 0)
      {
      Serial.print ("Found address: ");
      Serial.print (i, DEC);
      Serial.print (" (0x");
      Serial.print (i, HEX);
      Serial.println (")");
      count++;
      delay (1); 
      } 
  } 
  Serial.println ("Done.");
  Serial.print ("Found ");
  Serial.print (count, DEC);
  Serial.println (" device(s).");
} 

void loop() {}`

SvetoslavAsenov, 2019-05-25:
Fixed my issue by changing:

> Serial.begin (115200);

to

> Serial.begin (9600);

toxicpsion, 2019-06-23:
> I had to patch write method to make utf-8 chars and invalid chars prevent this crash:
> 
> ```
>         try:
>             self.output.write(text)
>         except UnicodeEncodeError:
>             self.output.write(text.encode("utf-8", "replace"))
> ```

worked for me, fixes problems with miniterm choking on boot garbage at 74880bps on 8266.

closed 2020-11-22 by stale[bot]