platformio/platformio-vscode-ide #35: forceUploadAndMonitor does not work on newer versions of VSCode

yaqwsx, 2017-08-17:
When I have "platformio-ide.forceUploadAndMonitor": true" I observe following behaviour:

1 Run VSCode
2. Upload - program is compiled, uploaded and serial monitor is started
3. Press upload again
4. Agree to terminate task
5. Get error: "The task '{0}' is already active. To terminate it use Terminate Task... from the Tasks menu."
6. Try upload again
7. Agree to terminate task
8. Get error: "Terminal with ID 2 does not exist (has it already been disposed?)"

I am using VSCode 1.15

sstaub, 2017-09-01:
Running tasks should **automatic** terminated when starting a new task.

ivankravets, 2018-02-02:
Should be fixed in the latest version. Please reopen if you still experience this issue

closed 2018-02-02 by ivankravets