platformio/platformio-vscode-ide #39: PIO Home screen comes up on every start

sstaub, 2017-09-01:
with v007 every time when you start the ide the pio home screen pops up, this is very annoying.
I'm missing "platformio-ide.showHomeOnStartup": false, in the config.

sstaub, 2017-09-01:
Sorry, i found the new button.

closed 2017-09-01 by sstaub

Shellwhale, 2019-01-13:
I am having the same issue, but I can't find the option. Where is it ?

brunomoreirafaro, 2019-04-11:
I think the default value for this option should be true

yhaiovyi, 2020-04-18:
^^^ Otherwise I call it spam.

muhammad-osaid-tekfocal, 2020-07-08:
Uncheck "show on Startup" on the very Home screen

sanbrother, 2020-09-29:
> Uncheck "show on Startup" on the very Home screen.
> 
> My current version (Core 4.3.4 Home 3.2.3):

Great! I don't know what 'the very Home screen' is, but after many tries, it works.

blackcoffeexbt, 2023-07-28:
> > Uncheck "show on Startup" on the very Home screen.
> > My current version (Core 4.3.4 Home 3.2.3):
> 
> Great! I don't know what 'the very Home screen' is, but after many tries, it works.

Right here for anyone else looking for the option.
[image]

Kamoba, 2023-12-22:
In my case unchecking "Show at startup" from home screen did not work, I had to make changes from ```Settings > Extensions > PlatformIO IDE```.
[image]

[image]