platformio/platformio-vscode-ide #40: Integrated debugger fails

DavidPowell, 2017-09-01:
On windows 10 x64, debugging fails under Visual Studio Code. I am logged into Visual Studio Code with my PlatformIO account. Pressing the debug button just causes the status bar to flash orange.

Board I am using is Nucleo-F103Rb. I am able to debug this same project under PlatformIO for Atom.

ivankravets, 2017-09-01:
Please click on "Terminal" button on the bottom toolbar (small icons). Try to start debugger from CLI
```
pio debug --interface gdb -x .pioinit
```

Does it work?

DavidPowell, 2017-09-02:
It gives the following output:

```
GNU ARM Eclipse 64-bits Open On-Chip Debugger 0.10.0-00114-g8419536 (2017-04-18-21:56)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : STLINK v2 JTAG v28 API v2 SWIM v17 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.242772
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection from pipe
Info : device id = 0x20036410
Info : flash size = 128kbytes
undefined debug reason 7 - target needs reset
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x00000000 in ?? ()
Temporary breakpoint 1 at 0x8005188: file src\main.cpp, line 229.
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x0800110c msp: 0x20005000
Loading section .text, size 0xcf9c lma 0x8000000
Loading section .ARM.exidx, size 0x8 lma 0x800cf9c
Loading section .data, size 0xa00 lma 0x800cfa4
Start address 0x80012dc, load size 55716
Transfer rate: 15 KB/sec, 9286 bytes/write.
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000003a msp: 0x20005000
Warn : keep_alive() was not invoked in the 1000ms timelimit. GDB alive packet not sent! (1715). Workaround: increase "set remotetimeout" in GDB
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x080012dc msp: 0x20005000
(gdb) target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x080012dc msp: 0x20005000
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x080012dc msp: 0x20005000
```
I'm not sure how relevant the libusb error is. I get the same output when running this command from the terminal in atom, but the atom debugger integration works fine.

ivankravets, 2017-09-09:
Does it work with the latest updates? Please `Menu: Tasks > Update packages...` and `Menu: Upgrade PIO Core`

DavidPowell, 2017-09-10:
Now when I run the debugger, it insists that I edit launch.json first. Then it fails with some error (I think something about path not being a valid string). Subsequently running debug just fails silently, so it's hard for me to reproduce the error message.

ivankravets, 2017-09-11:
Does debugging still work with Atom?

DavidPowell, 2017-09-11:
Yes, debugging is working fine under atom. I just tested it successfully with the mbed-blink example on a Nucleo F103RB.

ivankravets, 2017-09-11:
Could you share this ` launch.json`?

DavidPowell, 2017-09-12:
Here is the contents of `launch.json`

```
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "gdb",
            "request": "launch",
            "cwd": "${workspaceRoot}",
            "name": "PlatformIO Debugger",
            "target": "c:/Users/dap124/Documents/PlatformIO/Projects/170907-222728-mbed-blink/.pioenvs/nucleo_f103rb/firmware.elf",
            "gdbpath": "C:/Users/dap124/.platformio/penv/Scripts/piodebuggdb",
            "autorun": [ "source .pioinit" ]
        }
    ]
}
```
Trying again today, I noticed that if I ignore the warning and press debugging again, then debugging actually seems to work fine. So it seems that the issue may be fixed.

ivankravets, 2017-09-12:
Could you provide a screenshot of this warning? Thanks!

DavidPowell, 2017-09-12:
The tooltip shown appears when moving the mouse over the icon with the gear and the orange dot.

[image]

ivankravets, 2017-11-06:
This is OK, please ignore this warning.

closed 2017-11-06 by ivankravets