platformio/platformio-vscode-ide #41: CodeHelper maxes out to 100% in multiple processes

mjkkirschner, 2017-09-04:
I just installed platformio vscode extension and find that after building a few times using a teensy project that multiple CodeHelper processes are present in the activity monitor on Mac OS X and they are maxed out at 99 - 100%. After quitting VsCode they do not exit and must be force quit.

I have not seen this behavior previously using Arduino extension or regular vscode activity without extensions.

ivankravets, 2017-09-09:
Please check that you use the latest PIO IDE for VSCode and open PlatformIO IDE Terminal and type:
```
pio update
pio upgrade
```

Now, the better is to restart the machine or kill all `python/pioplus` processes.

ivankravets, 2017-11-03:
Please reopen if you still need help

closed 2017-11-03 by ivankravets

coddingtonbear, 2017-11-07:
I was about to reply that I was having this problem, too, but noticed that when running `platformio update`, I saw this message amongst the upgrade report:

```
Obsolete PIO Core v3.4.1 is used (previous was 3.5.0b2)
Please remove multiple PIO Cores from a system:
http://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system`
```

If future readers are bumping into this problem and happen across that error message -- if you follow the instructions listed there, it may fix the problem for you -- it did fix it for me.