platformio/platformio-vscode-ide #43: PlatformIO-IDE Installation Issue, "OSError: [Errno 2] No such file or directory:"

DeonMarais64, 2017-09-30:
I get "OSError: [Errno 2] No such file or directory:" installing from either "Atom" or "Visual Studio Code".

Installing from Atom :
OSError: [Errno 2] No such file or directory: '/cygdrive/c/Users/Deon-Marais/AppData/Local/atom/app-1.20.1/C:\\Users\\Deon-Marais\\.platformio\\penv'

Installing from Visual Studio Code
OSError: [Errno 2] No such file or directory: '/cygdrive/e/MSVSCode/C:\\Users\\Deon-Marais\\.platformio\\penv'

Any guidance or assistance would be appreciated.

fchaxel, 2017-10-06:
Hi,

Same problem, but I've found the solution  (after 3h of fight). 

On my side it's an installation bug due to python version when execution of virtualenv.tar.gz.

Inside the directory .platformio a good version of python.exe is present but during the installation if a path on another version of python is present then this one is used : if it's not a good version the installation could fails (for me because it was a GCC on Cygwin not MSC).

Open a console, launch python, if the version is not 2.17 MSC, then the bug is here.

Adds a path (at the begining of the path) on c:\users\youraccount\.platformio\python27
then restart VSCode. It will be OK. See it if never done https://www.computerhope.com/issues/ch000549.htm

So PlatformIO installer contains a bug : a call to the local python27\python should be done to avoid a wrong version.

Bye.

DeonMarais64, 2017-10-07:
Hi,
Thanks for the reply, It turns out I had a similar issue, In my case it was GCC on MingW32.
After some fiddling around, I eventually installed a complete version of Python2.7 (also adding the location to Path environment variable) before trying to install PlatformIO-IDE. After this installation went ahead smoothly. 
All the best.

ivankravets, 2018-01-16:
Thanks for the report! Version 0.10.1 is coming soon!

commit 860fe2e8 (2018-01-16): Ignore Python interpreter from Cygwin environment // Resolve #43
closed 2018-01-16 by ivankravets