platformio/platformio-vscode-ide #44: PlatformIO Home is not available

guoqingaa, 2017-10-18:
PlatformIO IDE as a vscode plug-in became difficult to use, Library can not download the local management, create a new project, in China can not access the PlatformIO Home page, the establishment of the project will fail, hoping to provide a sound offline version.

PlatformIO IDE作为vscode 插件之后变得难用了，Library不能本地下载管理，创建新项目时，在中国无法访问 PlatformIO Home 页，建立项目会失败，希望能提供完善的离线版本。

guoqingaa, 2017-10-18:
Can not open PIO Home: Can not start PIO master server: Error: timeout.
-> 无法打开“PIO Home”: Could not start PIO Home server: Error: timeout。

chall3ng3r, 2017-10-20:
+1

I am also getting same issue in Pakistan. "Unable to open 'PIO Home': Could not start PIO Home server: Error: timeout."

I believe this is the reason now VSCode extension is stuck on verifying PIO installation. https://github.com/platformio/platformio-vscode-ide/issues/45

chall3ng3r, 2017-10-21:
@guoqingaa, check pioplus.exe process in Windows firewall set to access public and private networks. 

I did it, and PIO Home starts working along with my stuck on "Verifying" issue.

ivankravets, 2017-11-03:
>  check pioplus.exe process in Windows firewall set to access public and private networks

Thanks a lot!

closed 2017-11-03 by ivankravets