platformio/platformio-vscode-ide #45: Verifying PlatformIO Core installation... stuck here.

chall3ng3r, 2017-10-20:
It was working perfectly with couple of VSCode updates back. But VSCode force updated, and the extension now is stuck here.

Have tried reinstalling, disabling with no use.

- VSCode Version: Code 1.17.2 (b813d12980308015bcd2b3a2f6efa5c810c33ba5, 2017-10-16T13:59:48.528Z)
- OS Version: Windows_NT ia32 10.0.16299
- Extensions: 

Extension|Author (truncated)|Version
---|---|---
scad|efb|1.0.1
cpptools|ms-|0.14.0
platformio-ide|pla|0.7.3
debug|web|0.21.2

---

Steps to Reproduce:

1. Install latest VSCode and PlatformIO Extention
2. Start VSCode. The status bar message with be "Verifying PlatformIO Core installation"

Reproduces without extensions: No

chall3ng3r, 2017-10-21:
Update, 

I researched on how PIO starts in VSCode, and found that couple of python.exe processes get stuck on "CLOSE_WAIT" connection status to an IP address. I killed both of these, and I noticed that PIO extension got pass the "Verifying" state to working state, and all command buttons appeared.

Also, pioplus.exe which is a local HTTP server (port 8010) starts when PIO Home is executed, was not setup properly in Windows firewall. This server renders PIO Home screen. After adding pioplus.exe process to firewall with public and private networks, PIO Home also showed up.

After restart of VSCode, now it seems to work even without killing those stuck python.exe processes. 

I guess main issue seem to be pioplus.exe not set correctly in firewall.

ivankravets, 2017-11-03:
Thanks a lot for your explanation how you resolved this issue!

closed 2017-11-03 by ivankravets

chall3ng3r, 2017-11-06:
Happy to contribute back :)

mamama1, 2019-02-09:
how about fixing this python crap? issue is still there, same behaviour.

chall3ng3r, 2019-02-09:
Uninstall Python on your PC, then download latest build of 2.7.x version series. Install it, it should fix the issue.

Make sure, after uninstall run python and see if you get error or python CLI, that means there's another version running.

I had two versions on my PC too, had to dig a little bit, but found it.

However, this still happens time to time. And only way for me is to kill those python processes under VSCode process, and restart VSCode.

sinnfeinn, 2020-02-25:
On mac closing python processes via activity monitor helped.