platformio/platformio-vscode-ide #46: Auto complete not working properly

ikorman, 2017-10-23:
HI

Last week I installed fresh installation of VSCode and platformio plug-in for it (using it for dev for ESP32 under Arduino FW). 

Everything works fine, except for auto complete function. For some reason, autocomplete always list all available functions, members and not only the ones related to given object. For example, when using Serial object from Arduino, I get hundred of suggestions for member functions/properties and not only the ones  related to Serial object.

I have installations running on two machines with same behavior. Fur curiosity reasons I tried Atom wit PIO and there it works fine.

Any suggestion what could be wrong here?

Thx.
Ivan

chall3ng3r, 2017-10-23:
I think its "by design" for VSCode extension. Same behavior for me too. I use VSCode as it's pretty snappy compared to Atom on my laptop (3rg gen i5).

joerneu, 2017-10-24:
It is working for me:

[image]

Check the VSCode C++ Extension - it had an update to support smarter Intellisense.

Set

`"C_Cpp.intelliSenseEngine": "Default"`

in your VSCode Settings.

Also, make sure to set:

`"intelliSenseMode": "clang-x64"`

in c_cpp_properties.json.

If not set to "clang" on Windows the F macro causes Intellisense errors.

See:

https://github.com/Microsoft/vscode-cpptools/blob/master/Documentation/LanguageServer/c_cpp_properties.json.md

The PlatformIO documentation should be updated.

chall3ng3r, 2017-10-24:
Thanks, just adding `"C_Cpp.intelliSenseEngine": "Default"` fixed it for me. Now listing on supported methods and properties.

ikorman, 2017-10-24:
Thx. I tried that, but still it remains like before. And also I can notices that intellisSenseMode is after restart of VSCode (after few seconds) reverting back to msvc-x64.

chall3ng3r, 2017-10-24:
For me, I didn't even have to restart VSCode, I just saved settings, and tried in open project, it worked.

joerneu, 2017-10-24:
@ikorman : Maybe recreating the Intellisense index does help?

Yes, there is an issue:

https://github.com/platformio/platformio-vscode-ide/issues/47

ikorman, 2017-10-24:
Also tried recreating the Intellisense index, as well I just removed and reinstalled VS Code and PIO. Still, effect is the same...

ivankravets, 2017-11-03:
Should be fixed in 0.7.4. Please update.

closed 2017-11-03 by ivankravets

ikorman, 2017-11-04:
Well, in the mean time I did find the cause and the fix it. There were two problems:

- after each IDE restarts "intelliSenseMode" reverted to "msVC-x64"
- there were some include files missing, which I had to add each time IDE reloaded (each time I had to add them manually again)

[image]

In short, I fix it it by adapting C_Cpp_ConfigurationProperties.js located it
[image]

- I changed the code so it always sets "clang-x64"
[image]

 - I added code that appends custome include paths each time IDE starts and c_cpp_properties.json is regenerated by reading one txt file in project folder

[image]

A nasty hack , but it did help me to solve the issues - is the fix somwhere in this area?

Thx.
Ivan

ivankravets, 2017-11-05:
It should work out of the box in 0.7.4 now. Could you confirm that? Thanks!

ikorman, 2017-11-05:
OK, I reverted to the original file in

[image]

and tried latest version. It works fine:

- clang-x64 remains after IDE restart
- there is no error regarding missing include paths

Great work :)

Now only to solve issue with naming/renaming projects and I'm happy :)

Thx.
Ivan