platformio/platformio-vscode-ide #47: Windows: Set "intelliSenseMode" property in c_cpp_properties.json file to "clang-x64"

joerneu, 2017-10-24:
PlatformIO should set `"intelliSenseMode"` property in `c_cpp_properties.json` file to `"clang-x64"`.

See:

https://github.com/Microsoft/vscode-cpptools/blob/master/Documentation/LanguageServer/FAQ.md

> How do I get the new IntelliSense to work with MinGW on Windows?

> Since MinGW is a relative of GCC, Microsoft mode compilation (which is the default on Windows) doesn't work very well with it. To use GCC/CLang mode, set the "intelliSenseMode" property in your c_cpp_properties.json file to "clang-x64".

Since PlatformIO uses CLANG like MinGW, Intellisense should use CLANG as well. (On Linux it is the default in VSCode C++ Extension.)

Manually editing `c_cpp_properties.json `does not work because PlatformIO overwrites the file and `"intelliSenseMode"` property returns to `"msvc-x64"` on Windows.

joerneu, 2017-10-28:
My workaround is to set "clang-x64" in the two template  files `c_cpp_properties.json.tpl` (Not sure why there are two files - it is probably enough to change just one of them.)

Search `c_cpp_properties.json.tpl` in folder:

`%USERPROFILE%\.platformio`

Change to:

```
...
            "defines": [
% for define in defines:
                "{{!define.replace('"', '\\"')}}",
% end
                ""
            ],
            "intelliSenseMode": "clang-x64"
...
```

Of course, an update to PlatformIO will overwrite this workaround until a permanent fix is available.

F-M-A, 2017-10-30:
That did the trick, but (dont really know if releated) now vscode does not detect anything releated with Arduino framework or AVR. For example "Serial" is marked as undefined but the autocomplete seems to work properly.

ivankravets, 2017-11-03:
Thanks a lot! Fixed in 0.7.4.

closed 2017-11-03 by ivankravets