platformio/platformio-vscode-ide #48: __AVR__ not set (in c_cpp_properties.json) -> Intellisense parsing incorrectly

lunds, 2017-10-25:
The __AVR__ macro does not seem to be defined when intellisense parses the code.
This breaks autocomplete etc. for code/libraries that use this macro (e.g. Low-Power library).

Project compiles fine though as gcc defines the macro.

In c_cpp_properties.json I find only the chip macro set (e.g. __AVR_ATmega32U4__), not the __AVR__ platform macro. Manually adding __AVR__ to the defines list makes intellisense resolve correctly. If this is the correct place for the define, I think it should be put there automatically.

Environment: 
Windows 7
VSCode settings: "C_Cpp.intelliSenseEngine": "Default"
c_cpp_properties.json: "intelliSenseMode": "msvc-x64"

ivankravets, 2017-11-03:
Could you try 0.7.4? Does it work?

lunds, 2017-11-04:
Thanks for the update :-)
I've now tried 0.7.4 with Core 3.5.0b2 but with the same result: `__AVR__` not defined in c_cpp_properties.json.

I digged a little in the commit log of pio core and found #b670ab4 (Initial support for non-ascii locales ) which touches the `__AVR_<build.mcu>__` macro. It changes the mcu related macro (which I think was ok before as well), but does not add the `__AVR__` platform macro.

My problem is that the library I'm testing (and presumably a number of other libraries as well - haven't checked) check on the generic `__AVR__` macro instead of (or in addition to) the more MCU specific macro.

Case in point, the "Low-Power" library (by Rocket ScreamElectronics Electronics), use the following structure to define the API depending on the Architecture/MCU in use:

```
#if defined (__AVR__)
	#if defined (__AVR_ATmega328P__) || defined (__AVR_ATmega168__) 
		...
	#elif defined __AVR_ATmega2560__
		...
	#endif

#elif defined (__arm__)
	...
#endif
```
and thus both AVR macros are required for autocomplete to work in this case.

ivankravets, 2017-11-04:
You need http://docs.platformio.org/en/latest/projectconf/section_env_build.html#build-flags

closed 2017-11-04 by ivankravets