platformio/platformio-vscode-ide #49: Feature: close open serial monitor before uploading

ikorman, 2017-11-09:
Here is one issue that annoys me quite a bit: if there is open serial monitor upload fails as given COM port can not be used. As I have terminal always open, I often forget to kill it before uploading the code and end up with error.

It would be very helpful that upload action preforms automatic closing of open serial monitor (should be configurable).

Thx.
Ivan

ivankravets, 2017-11-09:
We know about this issue and depend on https://github.com/Microsoft/vscode/issues/29855

A temporary solution is to use UPLOAD/SERIAL MONITOR buttons from bottom toolbar. In this case, we have control on the started tasks.

ikorman, 2017-11-09:
But his is how I use it now, I'm pressing those buttons on toolbar, but if the serial monitor is open, upload will fail with message error about COM port not being available, therefore I have to first kill (CTRL+C) serial monitor and then hit upload.

Thx.
Ivan

kristsm, 2017-12-16:
Yeah, I have the same issues? Are there any plans to fix this?

philomatic, 2018-06-14:
Same issue here, some more intelligence on this would be nice!

ivankravets, 2018-07-24:
It seems that you have multiple/conflicted extensions for VSCode. Please remove/disable the next extension for PlatformIO Project/Workspace:

1. Arduino
2. IoT Utility

Restart VSCode. Does it work now?

FeyverkSoft, 2018-07-24:
1) Arduino
it is not possible to disable/remove, it has become as a dependence on the platformio.
[image]  :(
2) IoT Utility not installed.

ivankravets, 2018-07-24:
Ha-ha. Is it a joke from Microsoft? See https://github.com/platformio/platformio-vscode-ide/blob/8d54e66dbc6fd6c29e472eb6792dddb990e02c0c/package.json#L552

We don’t depend on Arduino extension or Arduino IDE. 

Did you install this Arduino extension manually? Or, Microsoft installed it as addition to PlatformIO?

FeyverkSoft, 2018-07-25:
I'm manually installd the Arduino IDE extension, much earlier than PlatformIO extension.
This issues can be closed. I Removed all VS code settings, and installed it clean. Now everything works.
Thx.

ivankravets, 2018-07-25:
Reported to https://github.com/Microsoft/vscode/issues/55050

closed 2018-07-25 by ivankravets

thomas725, 2021-05-04:
I have the opposite problem: I'm using ArduinoOTA to upload firmware, and I'm annoyed that pressing the upload button automatically closes my serial monitor (which of course would make sense if I where uploading the firmware via serial connection, but it should check that's the case before closing the serial monitor)

commit e7b6c4f5 (2018-06-20): Automatically close Serial Port Monitor before uploading/testing // Resolve #49