platformio/platformio-vscode-ide #50: Workspace Question

DJManas, 2017-11-11:
Hello,

just a quick question. Are you supporting the new workspace feature of VSCode? I mean, I have 4 folders opened, but I want only to compile one of them. At the moment it is compiling the first one, but I want to compile e.g. 3rd one, because I want to do some memory optimization of few linked libraries.

Does it need some work from your side or is it M$’s bug? In atom it takes selected folder or active file and compiles the corresponding project.

Thanks,
Regards,
Petr Sourek

ivankravets, 2017-11-16:
We are working on this feature and will announce soon.

DJManas, 2017-11-16:
Great, if you need some beta testing, leave me a message here. I started to use VSCode, because I am not conmfortable with Atom and I have beta channel on.

electrek, 2018-02-16:
Any update on adding support for workspaces in VSCode (i.e. multiple folders)?

Thanks!

cranphin, 2018-03-18:
Just to be clear, this is to support this: https://code.visualstudio.com/docs/editor/multi-root-workspaces, correct? :)
It's the one thing I really miss from Atom :)

DJManas, 2018-03-18:
Yes. I can use command as a workaround, but I like that feature from Atom and finally it got into VSCode, which is much more faster than Atom...

smilexth, 2018-08-09:
Waiting selectable folder task  too

closed 2018-08-28 by ivankravets

DJManas, 2018-08-28:
Great thanks, this one works pretty well (quick test).

I get some issues on PlatformIO: Update platforms & packages / PlatformIO: Upgrade PlatformIO core to latest version like command doesn’t exists, but I will try to simulate it few times and create new issue.

DJManas, 2018-08-28:
Ok, the issue has something to do with PIO not fully initialized. Will try it in the next days. No issue for now.

commit 01de86d6 (2018-08-28): Support for multi-root workspaces; Implement Project Indexer and Task Manager // Resolve #50, Resolve #107