platformio/platformio-vscode-ide #52: Upgrading to 0.7.4 goes into an endless loop

proddy, 2017-11-25:
Windows 8 x64
VSCode 1.18.1

on starting VSCode it goes 'Verifying PlatformIO IDE..' then 'Installing PlatformIO' with output showing
"Installing PlatformIO Core...
Please don't close this window and don't open other folders until this process is completed."

on restart of VSCode the same sequence repeats. Verified on two separate installations with 0.7.3 previously installed.

VSCode Extension's shows PlatformIO IDE version 0.7.4 is correctly installed. Uninstalling and re-installing extension gives same result.

binderth, 2017-11-25:
same here:
[image]
Same goes with Atom as base, see the issues popping up there also: https://github.com/platformio/platformio-atom-ide/issues/1189

fcouceiro, 2017-11-25:
Same here! Any quick workaround yet?

ivankravets, 2017-11-25:
This issue was linked with Python's Arrow package ( https://github.com/crsmithdev/arrow/issues/495 ) and is already fixed.

Please remove `%HOME_DIR%/.platformio/penv` folder and restart IDE.

Sorry for the issue.

closed 2017-11-25 by ivankravets