platformio/platformio-vscode-ide #53: Unable to open 'PIO Home': Could not start PIO Home server: Error: timeout.

dp250f, 2017-11-25:
Attempted to install PlatformIO IDE (VSCode) on 2 computers.  Both are fresh installs, both give this error:
Unable to open 'PIO Home': Could not start PIO Home server: Error: timeout.

Both computers are Windows 10 1709 64-bit.  All I did was follow the instructions to install the VSCode version of PlatformIO IDE as provided on your website:
http://platformio.org/get-started/ide?install=vscode

Neither computer had previous versions of VSCode or PlatformIO installed, and I did restart after installing VSCode and after installing PlatformIO IDE extension.

The only integrated terminals available in VSCode are command shell and powershell (no python)
I notice these steps are commonly requested when troubleshooting this, so here they are:
```
C:\Users\dp250>ECHO %PATH%
C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Users\dp250\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;;C:\Program Files\Microsoft VS Code\bin

C:\Users\dp250>ECHO %PYTHONPATH%
%PYTHONPATH%
C:\Users\dp250>python --version'python' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\dp250>pip --version
'pip' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\dp250> pip list
'pip' is not recognized as an internal or external command,
operable program or batch file.
```

ivankravets, 2017-11-25:
This issue was linked with Python's Arrow package ( https://github.com/crsmithdev/arrow/issues/495 ) and is already fixed.

Please remove `%HOME_DIR%/.platformio/penv` folder and restart IDE.

Sorry for the issue.

rozpuszczalny, 2017-12-07:
Error still occurs on macOS 10.12.6 with Python 2.7.10, pip 9.0.1, platformio 3.5.0rc5, platformio-ide 0.8.2 and VSCode 1.18.1. Listing `~/.platformio/penv/lib/python2.7/sitepackages` show that arrow is still in 0.10 version.
```bash
$ ls
arrow                                 click                                 lockfile                              requests                              six.pyc
arrow-0.10.0-py2.7.egg-info           click-5.1.dist-info                   lockfile-0.12.2.dist-info             requests-2.18.4.dist-info             urllib3
bottle-0.12.13-py2.7.egg-info         colorama                              pip                                   semantic_version                      urllib3-1.22.dist-info
bottle.py                             colorama-0.3.9.dist-info              pip-9.0.1.dist-info                   semantic_version-2.6.0-py2.7.egg-info wheel
bottle.pyc                            dateutil                              pkg_resources                         serial                                wheel-0.30.0.dist-info
certifi                               easy_install.py                       platformio                            setuptools
certifi-2017.11.5.dist-info           easy_install.pyc                      platformio-3.5.0rc5-py2.7.egg-info    setuptools-38.2.4.dist-info
chardet                               idna                                  pyserial-3.4.dist-info                six-1.11.0.dist-info
chardet-3.0.4.dist-info               idna-2.6.dist-info                    python_dateutil-2.6.1.dist-info       six.py
```

Removing folder penv didn't help.

ivankravets, 2017-12-11:
@rozpuszczalny 
Please provide an output from a system Terminal (NOT IDE)
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

rozpuszczalny, 2017-12-11:
Output from Terminal (I've used Terminal app, not build-in terminal in VSCode)

<details>
<summary>Output of these command</summary>
<p>

```bash
$ echo $PATH
/Users/tomek/.sdkman/candidates/java/current/bin:/Users/tomek/.sdkman/candidates/groovy/current/bin:/Users/tomek/.sdkman/candidates/grails/current/bin:/Users/tomek/.rvm/gems/ruby-2.0.0-p598/bin:/Users/tomek/.rvm/gems/ruby-2.0.0-p598@global/bin:/Users/tomek/.rvm/rubies/ruby-2.0.0-p598/bin:/Users/tomek/activator/bin:/usr/local/mysql/bin:/Library/Frameworks/Python.framework/Versions/3.4/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/CrossPack-AVR/bin:/usr/local/share/dotnet:/usr/local/MacGPG2/bin:/Library/Frameworks/Mono.framework/Versions/Current/Commands:/Library/TeX/texbin:/Applications/Wireshark.app/Contents/MacOS:/Volumes/external/android-sdk/sdk/platform-tools:/usr/local/linaro/arm-linux-gnueabihf/bin:/Library/Frameworks/GStreamer.framework/Commands:/Volumes/external/android-sdk/sdk/tools:/Users/tomek/.rvm/bin
$ echo $PYTHONPATH

$ python --version
Python 2.7.10
$ pip --version
pip 9.0.1 from /Library/Python/2.7/site-packages (python 2.7)
$ pip list
altgraph (0.10.2)
arrow (0.12.0)
astroid (1.4.8)
autopep8 (1.2.4)
backports-abc (0.5)
backports.functools-lru-cache (1.2.1)
bcrypt (3.1.3)
bdist-mpkg (0.5.0)
bonjour-py (0.3)
certifi (2016.9.26)
cffi (1.10.0)
click (6.6)
configobj (5.0.6)
configparser (3.5.0)
docopt (0.6.2)
dronekit (2.9.0)
dronekit-sitl (3.2.0)
empy (3.3.2)
future (0.16.0)
ino (0.3.6)
isort (4.2.5)
Jinja2 (2.8.1)
lazy-object-proxy (1.2.2)
livereload (2.5.0)
lxml (3.6.4)
macholib (1.5.1)
Markdown (2.6.7)
MarkupSafe (0.23)
matplotlib (1.3.1)
MAVProxy (1.5.7)
mccabe (0.5.2)
mkdocs (0.16.1)
modulegraph (0.10.4)
monotonic (1.2)
nodemcu-uploader (0.3.1)
numpy (1.13.1)
ordereddict (1.1)
pep8 (1.7.0)
pip (9.0.1)
psutil (5.0.0)
py2app (0.7.3)
pyasn1 (0.1.9)
pycparser (2.18)
pylint (1.6.4)
pymavlink (2.0.9)
pyobjc-core (2.5.1)
pyobjc-framework-Accounts (2.5.1)
pyobjc-framework-AddressBook (2.5.1)
pyobjc-framework-AppleScriptKit (2.5.1)
pyobjc-framework-AppleScriptObjC (2.5.1)
pyobjc-framework-Automator (2.5.1)
pyobjc-framework-CFNetwork (2.5.1)
pyobjc-framework-Cocoa (2.5.1)
pyobjc-framework-Collaboration (2.5.1)
pyobjc-framework-CoreData (2.5.1)
pyobjc-framework-CoreLocation (2.5.1)
pyobjc-framework-CoreText (2.5.1)
pyobjc-framework-DictionaryServices (2.5.1)
pyobjc-framework-EventKit (2.5.1)
pyobjc-framework-ExceptionHandling (2.5.1)
pyobjc-framework-FSEvents (2.5.1)
pyobjc-framework-InputMethodKit (2.5.1)
pyobjc-framework-InstallerPlugins (2.5.1)
pyobjc-framework-InstantMessage (2.5.1)
pyobjc-framework-LatentSemanticMapping (2.5.1)
pyobjc-framework-LaunchServices (2.5.1)
pyobjc-framework-Message (2.5.1)
pyobjc-framework-OpenDirectory (2.5.1)
pyobjc-framework-PreferencePanes (2.5.1)
pyobjc-framework-PubSub (2.5.1)
pyobjc-framework-QTKit (2.5.1)
pyobjc-framework-Quartz (2.5.1)
pyobjc-framework-ScreenSaver (2.5.1)
pyobjc-framework-ScriptingBridge (2.5.1)
pyobjc-framework-SearchKit (2.5.1)
pyobjc-framework-ServiceManagement (2.5.1)
pyobjc-framework-Social (2.5.1)
pyobjc-framework-SyncServices (2.5.1)
pyobjc-framework-SystemConfiguration (2.5.1)
pyobjc-framework-WebKit (2.5.1)
pyOpenSSL (0.13.1)
pyparsing (2.0.1)
pyserial (3.0.1)
pysmb (1.1.18)
python-dateutil (1.5)
pytz (2013.7)
PyYAML (3.12)
scipy (0.13.0b1)
setuptools (18.5)
singledispatch (3.4.0.3)
six (1.10.0)
tornado (4.4.2)
uncompyle2 (1.1)
unpyclib (0.8.1)
vboxapi (1.0)
wheel (0.29.0)
wrapt (1.10.8)
xattr (0.6.4)
xortool (0.96)
zope.interface (4.1.1)
```
</p>
</details>

ivankravets, 2017-12-11:
@rozpuszczalny do you see any errors when run 
```
~/.platformio/penv/bin/pio home
```

rozpuszczalny, 2017-12-11:
Indeed, there is an error:
```bash
$ ~/.platformio/penv/bin/pio home
PlatformIO Plus (https://pioplus.com) v0.11.0
Error: Traceback (most recent call last):
  File "main.py", line 128, in main.main
  File "/Library/Python/2.7/site-packages/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Python/2.7/site-packages/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/Library/Python/2.7/site-packages/click/core.py", line 1057, in invoke
    Command.invoke(self, ctx)
  File "/Library/Python/2.7/site-packages/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Python/2.7/site-packages/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/Library/Python/2.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "main.py", line 69, in main.cli
  File "cmd_account.py", line 121, in cmd_account.cmd_check_permission
AssertionError

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to support@pioplus.com

============================================================
```

ivankravets, 2017-12-12:
@rozpuszczalny please download [a debug version](https://www.dropbox.com/s/kszrl5pmc29wfdd/tool-pioplus-121217.zip?dl=0) of `tool-pioplus` and replace it in `~/.platformio/packages` folder.

Try to run manually `pio home` and share here an output with Python system path (you will see it in console). Thanks!

rozpuszczalny, 2017-12-12:
And here goes the output
```
$ ./pio home
'/Users/tomek/.platformio/packages/pysite-pioplus', '/Users/tomek/.platformio/packages/tool-pioplus', '/Library/Python/2.7/site-packages/docopt-0.6.2-py2.7.egg', '/Library/Python/2.7/site-packages/lxml-3.6.4-py2.7-macosx-10.12-intel.egg', '/Library/Python/2.7/site-packages/dronekit_sitl-3.2.0-py2.7.egg', '/Library/Python/2.7/site-packages/six-1.10.0-py2.7.egg', '/Library/Python/2.7/site-packages/pymavlink-2.0.9-py2.7-macosx-10.12-intel.egg', '/Library/Python/2.7/site-packages', '/Users/tomek/.platformio/penv/lib/python2.7/site-packages', '/usr/local/lib/python2.7/site-packages', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python27.zip', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-darwin', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-mac', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/plat-mac/lib-scriptpackages', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-tk', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-old', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/lib-dynload', '/Users/tomek/Library/Python/2.7/lib/python/site-packages', '/Library/Python/2.7/site-packages', '/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python', '/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/PyObjC', '/Users/tomek/.platformio/packages/pysite-pioplus']
```

ivankravets, 2017-12-14:
@rozpuszczalny please run in PIO IDE Terminal `pio update`. Now, please restart IDE. Does it work now?

rozpuszczalny, 2017-12-14:
@ivankravets yes, now it works. Thank you! :)

ivankravets, 2017-12-14:
Thanks a lot for the report!

closed 2017-12-14 by ivankravets