platformio/platformio-vscode-ide #54: Arduino blink example project shows nonsense 'problems'

John-Titor, 2017-11-26:
The following canonical Arduino 'blink' example:

```
#include <Arduino.h>

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH); delay(1000);
  digitalWrite(LED_BUILTIN, LOW); delay(1000);
}
```

Set up for the Macchina M2 board (atmelsam) is plastered with error squiggles, due to each of the Arduino / Wiring function calls being marked with:

> file: 'file:///.../Documents/PlatformIO/Projects/171125-213209-macchina2/src/main.cpp'
> severity: 'Error'
> message: 'expression preceding parentheses of apparent call must have (pointer-to-) function type'
> at: '5,3'
> source: ''
> code: 'undefined'

Navigation to the definitions of the functions works, though they are also marked up with nonsense errors.

The problem appears to be due to this fragment in **.vscode/c_cpp_properties.json**:

>             "intelliSenseMode": "clang-x64",
>             "macFrameworkPath": [
>                 "/System/Library/Frameworks",
>                 "/Library/Frameworks"
>             ]

It looks like Visual Studio recently added a new "intelliSenseMode": "gccarm" mode, but my VS Code installation (1.18.1) rejects it.

ivankravets, 2017-11-28:
Could you provide your `platformio.ini`?

sean-mcmanus, 2017-11-28:
This appears to be a bug with the platformio-vscode-ide extension that is generating the c_cpp_properties.json and is missing required defines, such as in file `/.platformio/packages/framework-arduinosam/system/sam/CMSIS/CMSIS/Include/core_cm3.`h lines 82 to 98, where it checks the compiler defines for defining __ASM and __INLINE.

ivankravets, 2017-11-28:
We don't REDEFINE default macros from toolchains. Or better, we should not do that. I've never seen in embedded Make files when someone define `__GNUC__`.  Maybe, you should read default macros before starting a parse operation?

Why should define something which is already defined? Please reopen this issue https://github.com/Microsoft/vscode-cpptools/issues/1293#event-1362513626

We have 20+ different development platforms with different toolchains. We can't define FULL (including default macros for toolchain) macros list for VSCode. It will be HUGE.

sean-mcmanus, 2017-11-29:
Adding a define to the c_cpp_properties.json doesn't re-define it -- it makes the define visible to our language service compiler. Yeah, we can try to fix this via querying the compiler for it's defines like we do for the default includePath (we don't have an ETA on when this will be done though).

John-Titor, 2017-11-29:
platformio.ini contains:

```
> [env:macchina2]
> platform = atmelsam
> board = macchina2
> framework = arduino
```

ivankravets, 2017-11-29:
@John-Titor please update to 0.8.1. Should work now.

ivankravets, 2017-11-29:
@sean-mcmanus thanks a lot that took a look at this issue! 👍

commit 018159ee (2017-11-29): Add "__GNUC__" macro by default for VSCode // Resolve #54
closed 2017-11-29 by ivankravets