platformio/platformio-vscode-ide #57: pio invades every project/workspace

andig, 2017-12-03:
I'm using vscode for various coding tasks from php and node to arduino. The pio indexer/home screen comes up on every workspace, while only a few of those are actually arduino projects. This is very annoying especially when the indexer goes about looking at a few thousands of vendor depencies in php or node projects.

I'm looking for a way to disable pio by default and only enable on selected projects.

ivankravets, 2017-12-04:
Are you sure that PIO indexes every your project? See https://github.com/platformio/platformio-vscode-ide/blob/develop/src/main.js#L55

So, yes, we add PIO Home icon by default to any workspace but you can disabe it (uncheck "Show at startup").

Also, you can disable PlatformIO IDE extension for your other workspaces. Just navigate to `Extensions` and disable it.

andig, 2017-12-05:
> Are you sure that PIO indexes every your project?

I have not seen this behaviour again. I might have seen it still indexing another workspace when changing workspaces?

Not sure, but unchecking "home" solved the other problem, thank you :)

closed 2017-12-05 by andig