platformio/platformio-vscode-ide #58: Bug: PlatformIO incorrectly determines a user's Home folder and Documents folder

blujedi, 2017-12-06:
PlatformIO uses an incorrect (partially hard-coded) 'user home directory' and it also determines and handles the location of the Home Folder incorrectly. This causes major problems in (common) non-default situations.

Due to the details involved below explanation is rather long.

Windows does not have a well defined concept of 'User Home Directory'. It does have the concept of 'User folder" which is similar but not identical to a user's Home directory on Linux (a folder is a directory).

In recent Windows versions Microsoft is hiding the user from the concept of the 'User Folder' (probably because it is somewhat messy and not well defined). Windows is presenting the user special folder like  Documents, Pictures, Music, Desktop, Downloads etc. instead.

PlatformIO on Windows uses the User Folder as the user's 'Home directory' but incorrectly and part is hard-coded. In default situations this does not become visible but in (quite common) non-default situations you will run into a wall.

In recent Windows versions (Windows 8 and onward at least) there is a "Users" directory in the root of the C: drive. This is where WIndows stores User Folders.
I will use a sample male user with username _piouser_. He would have a User Folder C:\Users\piouser.

The User Folder contains at least the following:

- Standard 'Special Folders' like Documents, Pictures, Videos, Music, Downloads, Desktop etc.
- A hidden AppData directory with subdirectories Local, LocalLow and Roaming where the Windows OS and applications store user-specific data like application settings, temporary files etc.
- Hidden operating system files / copy of user profile, registry settings etc.
Unfortunately, many cross-platform applications - PlatformIO included - are not conforming to Windows guidelines for location of user related (configuration) data. Instead of storing it somewhere in the AppData tree they create a dot-prefixed directory name directly in the User Folder and store it there instead (like on Linux). PlatformIO creates its own .PlatformIO directory (e.g. C:\Users\piouser\.PlatformIO).
Applications that do not conform to Windows guidelines for location of user-related (configuration) data will often make their own - sometimes false - assumptions. PlatformIO included.

In practice it is often desired to separate storage of the operating system from storage of user data (documents, source code etc.') and put these on different disks. Reasons can be related to making of backups, easier repair of corrupt OS images, performance and can be related to the type and size of available disks.
Windows unfortunately does not allow a User Folder to be stored on a different disk than where the operating system is stored (C:). It cannot successfully cope with that.

**Quirk:**
It is possible to place special folders like Documents, Pictures, Video, Videos, Downloads, Desktop etc. in a different location on a different disk! This is supported by the Windows operating system.
This is done by changing the Location property of a special folder and then letting Windows move its contents.
IMPORTANT: When taking the Documents special folder as an example, when Documents is relocated/moved (by changing its Location property) to different location e.g. on a different drive, then it will be REMOVED from its old location in the User Folder and nothing will be put in its place as a place holder, not even a symbolic link. So it is not possible to determine the new location of Documents by inspecting the User Folder, nor do any environment variables exist to assist in that. The current location of the Documents folder (and other special folders) can only reliably be determined by quering the operating system via Windows API's.

_(I always create a directories D:\Users for separating user data storage from OS storage. In that directory I create a 'Home folder' for myself (e.g. D:\Users\piouser) and then relocate al specials folders to there (e.g. D:\Users\piouser\Documents.)_

If directory C:\Users\piouser\Documents is missing this should alarm an installer that the situation is not default and indicates that the Documents folder has been relocated. An application should never make it's own assumptions and *always* query the operating system for the actual location of standard Special Folders.
But the PlatformIO installer does NOT. If PLatformIO finds C:\Users\piouser\Documents missing then it will just create a C:\Users\piouser\Documents (ouch!) and create a PlatformIO subdirectory (with Projects subdirectory) in the self-created Documents directory. This is bad and leads to problems in (common) non-default situations.

**Quirk:**
User Folders are located on the C: drive in C:\Users but there is also the concept of 'Home folder'.
'Home folder' is the directory pointed to by a combination of the environment variables HOMEDRIVE and HOMEPATH. 
HOMEDRIVE contains only the drive and HOMEPATH contains the path relative to the root of the drive (but not the drive).does not include the drive itself). In Linux ~ refers to the full path of the user's Home directory. Not so in Windows (and there is nothing comparable to ~ on Linux).

**Quirk:**
'Home folder' and therefore the values of HOMEDRIVE and HOMEPATH are determined by the value of: 'User properties/Profile tab/Home folder/Local path'. This value is empty by default which means that Home folder points to the same location as User Folder but they can also be different and Home folder can even point to a different drive.

**Quirk**
Changing the value of Home folder/Local path does effect the values of HOMEDRIVE and HOMEPATH but it does not have any effect on (the use of) the User Folder, nor does it have any effect on standard Special Folders, neither in their default location nor when relocated.

So it turns out that User Folder, Home Folder and Special Folders are inconsistent and they can all point to different locations and different drives which does not make things easier.

**PROBLEMS with PlatformIO:**

PlatformIO uses Home Folder as the 'user's root folder' which is displayed in file dialogs under 'Places' as the user's username.

Problem 1:
PlatformIO uses only HOMEPATH for determining the Home Folder location but *not* HOMEDRIVE and uses a hard-coded "C:" for HOMEDRIVE instead.

Problem 2:  
PlatformIO assumes that the Documents folder is a subdirectory of the Home Folder, which is true only in default situations but not in many common non-default situations.
PlatformIO should use Windows API's to query for the actual location of the Documents special folder instead.

Problem 3:
When PlatformIO does not find the Documents folder where it expects it then it creates a Documents folder there itself (bad!). PlatformIO should never touch the Documents Folder itself and leave that to the operating system.

**HOW TO FIX:**

1) Determine the actual location of the Documents special folder via appropriate Windows API's (and DON'T make any assumptions yourself.
Notes: The location of Documents can be different from User Folder and different from Home Folder (meaning it does not have to be a subdirectory of them) and can even be located on a different drive.
The (relocated) location of the Documents folder is technically unrelated to the location(s) of User Folder and Home Folder.

2) Use a combination of both the HOMEDRIVE and HOMEPATH environment variables for determining the correct full path to the Home Folder.
Notes: The Home Folder can point to the same location as User Folder but it can also be different and even point to a different drive.

ivankravets, 2017-12-12:
We don't use a hardcoded path for the home directory. We use https://docs.python.org/2.7/library/os.path.html#os.path.expanduser

Please set correctly HOME and USERPROFILE environment variables.

blujedi, 2017-12-13:
### Please reopen the issue. The issues (bugs) still exist.
_I do not understand why you have closed the issue without verifying the details I have provided and without applying the provided reproduction steps._

You are referring to a Phython function that PlatformIO is using but only mentioning that does not solve any of the issues that I am addressing. It is even possible that part of the issues is caused by that Python function. You should VERIFY my description above and VERIFY the issues by applying the reproduction steps.

In response to your post,

### First

> Please set correctly HOME and USERPROFILE environment variables.
- NO: Windows is NOT LINUX! There  is no HOME environment variable in Windows and there should not be! In Windows you should use a combination of HOMEDRIVE and HOMEPATH instead!
- NO: USERPROFILE is a FIXED operating system location. Users should never change the USERPROFILE variable.

### Second

> We don't use a hardcoded path for the home directory. We use https://docs.python.org/2.7/library/os.path.html#os.path.expanduser

says:
> _os.path.expanduser(path)_
>On Windows, HOME and USERPROFILE will be used if set, otherwise a combination of HOMEPATH and HOMEDRIVE will be used. An initial ~user is handled by stripping the last directory component from the created user path derived above.

- "HOME": There is NO HOME environment variable on Windows and there should not be!
- "and USERPROFILE will be used if set": The USERPROFILE environment variable **is ALLWAYS set** (but not listed) and the user should **never** change it!
- "otherwise a combination of HOMEPATH and HOMEDRIVE will be used": 
Well, that maybe is what it SHOULD do but it DOES NOT. 
**This is a BUG and is why I created this issue.**
- "An initial ~user is handled by stripping the last directory component from the created user path derived above": **An 'initial' user should be 'stripped' from USERPROFILE path ONLY**, NOT from Home Folder (HOMEDRIVE + HOMEPATH) or anything else!
_(FYI from USERPROFILE and not from USERNAME because changing the username will not change the USERPROFILE location.)_

And WORSE: PlatformIO blatantly creates a Documents folder in UserProfile when the Documents Folder is not in it's default location. 
The reason is that PlatformIO does NOT query Windows for the correct Documents location and PlatformIO makes its own incorrect assumptions instead. That is *really BAD!*
**This is a BUG and is why I created this issue.**

## To fix the described issues this is what PlatformIO should do

- Use the **USERPROFILE** environment variable to determine the location of the User Profile.
The UserProfile is the location where the **.PlatformIO** directory should be stored.

- **Use the SHGetKnownFolderPath Windows API with FOLDERID_Documents to determine the location of the user's Documents Folder.** 
This is the only correct and future proof way to determine the location of the Documents folder!
_(The location of the Documents Folder CANNOT be determined from ANY environment variables!
Don't make 'your own assumptions' and stick to the Windows requirements and guidelines.)_
The Documents folder is where the **PlatformIO** and **PlatformIO\Projects** directories have to be placed. 

- Use a combination of **HOMEDRIVE** and **HOMEPATH** to determine the location of the *Home Folder*. The Home Folder is what will be the current directory when you open a command prompt on Windows. The Home Folder is also what should be opened when you select 'Places/<useraccount name> in the PlatformIO IDE 'Open Project'

### Avoid these Pitfalls
Note the facts below and prevent any pitfalls by making 'your own' (incorrect) assumptions:
- Only in a default configuration do USERPROFILE and Home Folder to the same location!
- Only in a default configuration is the Documents Folder a subdirectory of the USERPROFILE!
- The Documents Folder location is independent from the USERPROFILE location!
- The Documents Folder location is independent from the Home Folder location!
_(It is most practical when the Documents Folder is a subdirectory of the Home Folder but is is not required.)_
The only reliable way to determine the Documents Folder location is by using the SHGetKnownFolderPath Windows API (or a higher level API that uses it).
_While it is possible to find the same information in the registry, but getting the information from the registry directly is not preferred and not future proof._
- The Documents Folder location CANNOT be determined from environment variables.

Attached is the file GetUserFolders.cpp.txt _(remove the .txt extension after downloading, I was not able to upload it with .cpp extension)_. This is a simple example in C# that properly determines and lists the correct UserProfile, Documents Folder and Home Folder locations.
_To keep the example simple it uses higher level .NET API's which abstracts the underlying SHGetKnowFolderPath API._

The example can be compiled to an executable with `"C:\Windows\Microsoft.NET\Framework\v4.0.30319\csc" GetUserFolders.cpp`
And then execute the generated GetUserFolders.exe

[GetUserFolders.cpp.txt](https://github.com/platformio/platformio-vscode-ide/files/1555632/GetUserFolders.cpp.txt)

ivankravets, 2017-12-13:
1. Please provide an output from `python -c "import os; print os.path.expanduser('~')"`
2. When creating new project please uncheck "Default location" and you will be able to use custom project folder.

blujedi, 2017-12-13:
1. Please provide an output from python -c "import os; print os.path.expanduser('~')"

D:\Users\PioUser>getuserfolders
UserProfile path is:      C:\Users\ PioUser
Home folder path is:      D:\Users\ PioUser
Documents folder path is: D:\Users\ PioUser \Documents

D:\Users\ PioUser>python -c "import os; print os.path.expanduser('~')"
C:\Users\ PioUser

2. When creating new project please uncheck "Default location" and you will be able to use custom project folder.

Really?
That would be a user experience hell!

It is PlatformIO that needs to be fixed, not its users.

blujedi, 2017-12-13:
Thanks for re-opening the issue.

ivankravets, 2017-12-13:
1. As you can see, the issue with Python. PlatformIO is written in Python and we use its API

2. No need to fix something. See screenshot below

[image]

ivankravets, 2017-12-13:
Could I close this issue?

blujedi, 2017-12-13:
No of course you cannot close this issue.
Your suggestion is a pure user experience hell.

You should get it fixed in PlatformIO !
If this is a Python issue then contact the Python team and have them fix it
and otherwise try to search for alternative please.

ivankravets, 2017-12-13:
1. We use user's HOME directory, not documents folder.
2. Python handles correctly home directory, see "On Windows, HOME and USERPROFILE will be used if set, otherwise a combination of HOMEPATH and HOMEDRIVE will be used"
3. If you would like to store projects in a custom location, use GUI for that.

In any case, we have separate issue which will allow to override default location for projects https://github.com/platformio/platformio-core/issues/1161

blujedi, 2017-12-13:
Dammit, Documents folder is WINDOWS STANDARD for user 'documents'.

ivankravets, 2017-12-13:
Please provide an output of `python -c "import os; print os.environ"`

blujedi, 2017-12-13:
I am not going to dump that fully here (privacy).
What entries are you specifically interested in?

ivankravets, 2017-12-13:
Do you see env var which is related to `Documents folder path is: D:\Users PioUser\Documents`?

blujedi, 2017-12-13:
No, but that is what I already explained in my descriptions above:

### Avoid these Pitfalls
Note the facts below and prevent any pitfalls caused by making 'your own' (incorrect) assumptions:
- The Documents Folder location CANNOT be determined from environment variables, unfortunately.

(Thanks Microsoft...)

riks5, 2018-08-21:
Hi,
I found a easy solution for eclipse ide.
See attached picture.

[image]

prismv, 2019-03-15:
When using Linux I can confirm that the PlatformIO extension in VSCode always tries to use %HOME%/Documents/PlatformIO/Projects. There is apparently no way to redefine this as a user as a general setting, which is quite annoying if one wants to use a custom directory structure in the HOME folder. I try to use a simple link %HOME%/Documents/PlatformIO/Projects to the project folder I would like to use, but the link was overwritten when a new project was created.

ivankravets, 2019-03-15:
Have you tried to overwrite it via http://docs.platformio.org/en/latest/userguide/cmd_settings.html#projects-dir ?

prismv, 2019-03-15:
Okay, it was not clear that the general pio settings are working here. Yes, the projects folder is also changed in vscode ide.
Thanks.

Galbi3000, 2019-04-17:
This is still an issue.

I use a limited sized SSD for my system drive and have an HDD for non-essential programs and data, that includes my Documents folder. I would prefer to have the Projects folder in that Documents folder, not on the system drive. Doing lots of development work on an SSD drastically reduces the write cycle lifespan of the drive!

My home directory is C:\users\galbi but my Documents folder is at D:\Documents.

To reproduce the problem yourself open an Explorer window using WIN+E, this should open the window at the "This PC" location. At the top you have the user folders. Right click Documents and select properties. There should be a "Location" tab, go to that. Now change the location of the Documents folder from there (the "Move..." button will move existing files to the new location which can take time).

The correct way to get that location is reading it from the registry: HKEY_CURRENT_USER\Software\Microsoft\Windows\CurrentVersion\Explorer\Shell Folders\Personal

You should never assume it's in the default location off of the user home directory.

A quick Google on how to read the registry in Python reveals it can be done using the _winreg library. I'm new to Python so I can not give further details but I am sure you know how to obtain and use the library.

Alternatively you could allow manual setting of the PlatformIO folder, or even just the Projects folder :)

blujedi, 2019-04-17:
No, the registry is not the correct way to get these settings!
Not future proof therefore error prone. Exactly what is not wanted.

The only CORRECT way to get these settings is by calling the Windows API.

I spent quite some time on it after opening the issue and I even did write some sample Python code to call the relevant Windows API but I have not bothered since, because there is a workaround.

The workaround was already mentioned by @ivankravets above: Use the **projects_dir** setting to set a (global) custom projects directory.

While the workaround works, it is ugly that one has to needed because a poor implementation of an essental feature in PlatformIO. (More details and background can be found in the top post.)

ivankravets, 2019-04-17:
We plan to add GUI for `pio settings` to PIO Home soon. So, you will be able to tune all settings from PIO Home.

blujedi, 2019-04-17:
It is nice to have a GUI for `pio settings` but that does still does not address the core of the issue (incorrectly automatic determination of where a Windowss users' PlatformIO project directory should be located, while it _is_ possible to do it correctly automatically).

Galbi3000, 2019-04-17:
> 
> 
> No, the registry is not the correct way to get these settings!
> Not future proof therefore error prone. Exactly what is not wanted.
> 
> The only CORRECT way to get these settings is by calling the Windows API (as described in the top post).
> 
> I spent quite some efforts on this after opening the issue and I even did write some sample Python code to call the relevant Windows API but I have not bothered since (because there is a workaround).
> 
> The workaround was already mentioned by @ivankravets above: Use the **projects_dir** setting to set a (global) custom projects directory.
> 
> While the workaround works, it is ugly that one needs to use it for something that can be fixed in PlatformIO itself.

The Windows API is just as subject to change as the registry so it's not future proof either, nothing is in the programming world. Just take a look at the API function you quoted, SHGetKnownFolderPath, it has gone through at least 3 name changes to get to how it's named now!

The Windows API also gives software access to the registry which is where software is suposed to store settings, that's what the registry was designed for, to replace the multitude of .ini files. So accessing the registry directly is allowed making using the API not the only correct way.

As Platform IO is written in Python for cross platform compatibility, used in VSC on Linux, MacOS and Windows, this issue is made more complex. I assume it would be easier to just read the registry directly in the place of using environment variables in Windows than to try and make it use the Windows API. As I said, I am new to Python so it's beyond my scope. But I am not new to Windows API and programming for Windows using C++, I've been doing it since the days of Windows 3.1, professionally until 2000 and on & off since then as one of many programming hobbies :)

blujedi, 2019-04-17:
Still, using the registry is just bad practice. API is public interface, registry is implementation details.

Api access from Python is possible. I have done it when I opened this issue.
Still have it somewhere. So should not be a problem to do in PlatformIO.

Galbi3000, 2019-04-17:
Well, we have given them the info on how to fix it, whether or not they impliment either way is yet to be seen but looking at the time span between your OP and today where it's still an issue it looks like they just do not take it seriously enough. But it is serious for users like me who do not want to do compiling and development work on an SSD which has a finite number of times it can be written to.

qvdm, 2019-07-08:
I just came here after spending hours to try and fix the problems created by this issue and was astonished to read that it has been a known issue since 2017.  

In my case, I use OneDrive, which moves the Documents folder (and references to it) to inside the OneDrive folder.  Every_single_one of my windows applications know how to deal with this, by just following Windows recommended practices, EXCEPT for PIO.  

The fact that this issue keeps getting closed without a fix reflects very poorly on the quality of the Platformio project.

ivankravets, 2019-07-08:
@qvdm Have you read all comments in this issue? If no, please re-read. The issue is very specific to your OS. PlatformIO is cross-platform tools and Windows is one of the OS which it supports. We use Python API to get the user home directory. There is a setting which you can override if don't like default value. See http://docs.platformio.org/en/latest/userguide/cmd_settings.html#projects-dir

 > very poorly on the quality of the Platformio project

Do you have enough software/architect expertise to say that?

P.S: This is free and open source project. I recommend helping improve it instead of claiming that this  is "very poor and low-quality project".

ivankravets, 2019-07-08:
People in this thread, could you open your system terminal and type command below:
```
python -c "import ctypes.wintypes; buf = ctypes.create_unicode_buffer(ctypes.wintypes.MAX_PATH); ctypes.windll.shell32.SHGetFolderPathW(None, 5, None, 0, buf); print(buf.value)"
```

Do you see a valid value for your documents directory?

qvdm, 2019-07-08:
The command above reports the correct location of the documents directory for me (i.e. not the location that Platformio uses).

ivankravets, 2019-07-08:
Resolved in https://github.com/platformio/platformio-core/commit/ffcf6b873a5f16f843c991394a82ebfbbd4f3c3c

Please re-test with upcoming PlatformIO Core 4.0 => https://community.platformio.org/t/platformio-core-4-0-beta/8304

closed 2019-07-08 by ivankravets