platformio/platformio-vscode-ide #60: Could not start PIO Home server: Error: timeout。

chenjingyuanku, 2017-12-13:
When I upgraded to version 0.8.2, I was unable to open PIO HOME.
I removed `%HOME_DIR%/.platformio/penv` folder and restart IDE.
It's useless.
I have tried reinstalling, but it still prompts "Could not start PIO Home server: Error: timeout。"

ivankravets, 2017-12-13:
Please open PIO IDE Terminal and type `pio home`, do you see any errors?

chenjingyuanku, 2017-12-13:
Traceback (most recent call last):
  File "main.py", line 12, in init main
  File "C:\Users\Y\.platformio\penv\lib\site-packages\click\__init__.py", line 18, in <module>
    from .core import Context, BaseCommand, Command, MultiCommand, Group, \
  File "C:\Users\Y\.platformio\penv\lib\site-packages\click\core.py", line 8, in <module>
    from .types import convert_type, IntRange, BOOL
  File "C:\Users\Y\.platformio\penv\lib\site-packages\click\types.py", line 5, in <module>
    from ._compat import open_stream, text_type, filename_to_ui, \
  File "C:\Users\Y\.platformio\penv\lib\site-packages\click\_compat.py", line 6, in <module>
    from weakref import WeakKeyDictionary
  File "C:\Users\Y\.platformio\penv\lib\weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref

ivankravets, 2017-12-13:
It seems that you have old version of Python in a system. Please navigate to OS Control Panel and remove ALL pythons, then remove `C:\Users\Y.platformio\penv` and `C:\Users\Y.platformio\python2` folders if they exist. Restart IDE.

chenjingyuanku, 2017-12-14:
The problem is solved，thank you so much!

closed 2017-12-14 by chenjingyuanku

withkittens, 2018-01-22:
Are there any other workarounds without removing other Python installations?

ivankravets, 2018-01-23:
@withkittens I've just removed all Pythons from VM, install Python 3.6.4, then started PIO IDE and all works good.

Could you remove ALL Pythons, install Python 3.6 again and restart PIO IDE? Also, don't forget to remove "C:\Users\withkittens\.platformio"

lzd-1230, 2022-12-15:
Hi @ivankravets, I have found solution for maybe majority guys from China, it is the network problem. 

Background information: I've start to install platformio-vscode-plugin from a whole new computer system(both windows and linux) and it all get this error

And i follow the instruction using `pio home` and it couldn't install the dependency package it just stuck here:
[image]

and I consider that is the **network problem**(which is resonable for both system doesn't work).

because vscode seems doesn't use proxy by default

So the solution is pretty clear:  using global proxy for example **tun mode for clash**

after global proxy, we can successfully run `pio home` in this cli
[image]

and then restart vscode you can successfully using it!!!

thanks for the inspiration from: https://github.com/platformio/platformio-vscode-ide/issues/205#issuecomment-1233183495

ivankravets, 2022-12-18:
@lzd-1230 have you tried to configure proxy using https://docs.platformio.org/en/latest/integration/ide/vscode.html#proxy-server-support ?

lzd-1230, 2022-12-19:
> @lzd-1230 have you tried to configure proxy using https://docs.platformio.org/en/latest/integration/ide/vscode.html#proxy-server-support ?

Thanks for mention that, I thought it would help. I've try once, when installing platformio plugin with vscode proxy, it doesn't help after installing and when open platformio ,it still stucked. But maybe proxy in platformio would help, It's the key that this problem is cause by network problem, so maybe there's many ways to solve it.