platformio/platformio-vscode-ide #61: Unable to install platform espressif32

tekk, 2017-12-14:
```>pio platform install espressif32
Please wait while upgrading PlatformIO...
Platform Espressif 32
--------
Updating espressif32                     @ 0.11.0         [Up-to-date]
Updating tool-esptoolpy                  @ 1.20100.0      [Up-to-date]
Updating framework-arduinoespressif32    @ 1.3.2          [Up-to-date]
Uninstalling toolchain-xtensa32 @ 2.50200.0:    [OK]

PlatformIO has been successfully upgraded to 3.5.0rc6!

*************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for IoT development > http://platformio.org/platformio-ide
- support us with PlatformIO Plus > https://pioplus.com
*************************************************************************************************************************************

PlatformManager: Installing espressif32
espressif32 @ 0.11.0 is already installed
PackageManager: Installing toolchain-xtensa32 @ ~1.50200.2
Downloading  [------------------------------------]    0%
Please manually remove file `C:\Users\tekks\.platformio\packages\_tmp_installing-au4eow-package\toolchain-xtensa32-windows-1.50200.2.tar.gz`
Warning! Package Mirror: [Error 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\tekks\\.platformio\\packages\\_tmp_installing-au4eow-package\\toolchain-xtensa32-windows-1.50200.2.tar.gz'
Looking for other mirror...
Downloading  [------------------------------------]    0%
Please manually remove file `C:\Users\tekks\.platformio\packages\_tmp_installing-zzgqel-package\toolchain-xtensa32-windows-1.50200.2.tar.gz`
Warning! Package Mirror: [Error 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\tekks\\.platformio\\packages\\_tmp_installing-zzgqel-package\\toolchain-xtensa32-windows-1.50200.2.tar.gz'
Looking for other mirror...
Error: Could not install 'toolchain-xtensa32' with version requirements '~1.50200.2' for your system 'windows_amd64'.
If you use Antivirus, it can block PlatformIO Package Manager. Try to disable it for a while.```

ivankravets, 2017-12-14:
What are your OS version and locale?

ivankravets, 2017-12-14:
This a VSCode Terminal bug. I played a few hours on a remote machine (some of our users provided remote access) with this issue yesterday. The problem is linked with Python Socket library.  I don't know how does it relate to VSCode Terminal extension but it raises `IOError: [Errno 0] Error` exception with a simple Python script:

```python
from urllib2 import urlopen

url = "http://dl.platformio.org/packages/toolchain-xtensa-windows-1.40802.0.tar.gz"

r = urlopen(url)
i = 0
for chunk in r.read(1024):
    i += 1
    print i
```

If you run this script on an affected machine using native CMD terminal or even our Terminal for Atom, there is no issue.

I have a few VM machines with different Window versions and can't reproduce it. 

We will provide a step-by-step guide in our docs for VSCode in "Know issues" section and explain how to use installed PIO Core with native `cmd` tool for the first time (to download dependent toolchains).

Related issues: https://community.platformio.org/t/packagemanager-is-unable-to-install-tool/2662

---------

@Tyriar do you have any ideas? We experience this problem a long time. I know you use `pty.js` the same as we do in Atom, why it works with Atom but does not work with VSCode?

ivankravets, 2017-12-14:
Docs is updated: http://docs.platformio.org/en/latest/ide/vscode.html#known-issues

Tyriar, 2017-12-14:
@ivankravets VS Code now uses [node-pty](https://github.com/Tyriar/node-pty) which has diverged significantly from pty.js, maybe the problem is in one of the changes?

ivankravets, 2017-12-14:
@Tyriar would you like to have access to the affected machine and debug this issue? I can agree that with our user.

Thanks!

tekk, 2017-12-14:
Win10 x64, locale sk_SK.

--------------------------
Ing. Peter Javorský

tekk, 2017-12-14:
> unable-to-install-tool/2662

Look at the logs. It states: cannot access file, other process is using
it.. and there are 2 temporary direcories created in which both the pio
wanted to download xtensa framework.

--------------------------
Ing. Peter Javorský

ivankravets, 2017-12-14:
@tekk please open PIO IDE Terminal in VSCode and type `pio upgrade`. You should be upgraded to 3.5.0rc7. Try to build a project again. Which errors do you see now?

Tyriar, 2017-12-14:
@ivankravets I think this class of issues can probably be reproduced easily, I don't have time to investigate atm though unfortunately.

tekk, 2017-12-14:
```
>pio upgrade
Updating contrib-piohome                 @ 0.5.0          [0.5.3]
UUninstalling ?[36mcontrib-piohome?[0m @ 0.5.0:
[OK]
CorePackageManager: Installing contrib-piohome @ 0.5.3
Downloading  [------------------------------------]    1%
Please manually remove file `C:\Users\tekks\.platformio\packages\_tmp_installing-algpys-package\contrib-piohome-0.5.3.tar.gz`
Warning! Package Mirror: [Error 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\tekks\\.platformio\\packages\\_tmp_installing-algpys-package\\contrib-piohome-0.5.3.tar.gz'
Looking for other mirror...
Downloading  [------------------------------------]    1%
Please manually remove file `C:\Users\tekks\.platformio\packages\_tmp_installing-ysrawq-package\contrib-piohome-0.5.3.tar.gz`
Warning! Package Mirror: [Error 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\tekks\\.platformio\\packages\\_tmp_installing-ysrawq-package\\contrib-piohome-0.5.3.tar.gz'
Looking for other mirror...
Error: Could not install 'contrib-piohome' with version requirements '0.5.3' for your system 'windows_amd64'.
If you use Antivirus, it can block PlatformIO Package Manager. Try to disable it for a while.
```

tekk, 2017-12-14:
```
>pio upgrade
Updating tool-pioplus                    @ 0.11.0         [0.12.0]
Uninstalling tool-pioplus @ 0.11.0:     Please manually remove file `C:\Users\tekks\.platformio\packages\tool-pioplus\pioplus.exe`
Error: Traceback (most recent call last):
  File "c:\users\tekks\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\tekks\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\tekks\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\tekks\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\tekks\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\tekks\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\tekks\.platformio\penv\lib\site-packages\platformio\commands\upgrade.py", line 30, in cli
    update_core_packages(silent=True)
  File "c:\users\tekks\.platformio\penv\lib\site-packages\platformio\managers\core.py", line 89, in update_core_packages
    pm.update(name, requirements, only_check=only_check)
  File "c:\users\tekks\.platformio\penv\lib\site-packages\platformio\managers\core.py", line 51, in update
    result = PackageManager.update(self, *args, **kwargs)
  File "c:\users\tekks\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 735, in update
    self.uninstall(pkg_dir, trigger_event=False)
  File "c:\users\tekks\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 676, in uninstall
    util.rmtree_(pkg_dir)
  File "c:\users\tekks\.platformio\penv\lib\site-packages\platformio\util.py", line 650, in rmtree_
    return rmtree(path, onerror=_onerror)
  File "C:\Python27\Lib\shutil.py", line 252, in rmtree
    onerror(os.remove, fullname, sys.exc_info())
  File "c:\users\tekks\.platformio\penv\lib\site-packages\platformio\util.py", line 648, in _onerror
    raise e
WindowsError: [Error 5] Access is denied: 'C:\\Users\\tekks\\.platformio\\packages\\tool-pioplus\\pioplus.exe'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

ivankravets, 2017-12-14:
@tekk please close IDE and remove `c:\users\tekks\.platformio\penv` folder manually. Start IDE. Please provide build output.

tekk, 2017-12-14:
> Installing PlatformIO Core...
> Please don't close this window and don't open other folders until this process is completed.

...
```
> Executing task: C:\Python27\Scripts\platformio.exe run <

*************************************************************************************************************************************
If you like PlatformIO, please:
-Error: Traceback (most recent call last):
  File "c:\python27\lib\site-packages\platformio\__main__.py", line 107, in main
    cli(None, None, None)
  File "c:\python27\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\python27\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\python27\lib\site-packages\click\core.py", line 1024, in invoke
    Command.invoke(self, ctx)
  File "c:\python27\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\python27\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\python27\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\python27\lib\site-packages\platformio\__main__.py", line 76, in cli
    maintenance.on_platformio_start(ctx, force, caller)
  File "c:\python27\lib\site-packages\platformio\maintenance.py", line 49, in on_platformio_start
    after_upgrade(ctx)
  File "c:\python27\lib\site-packages\platformio\maintenance.py", line 146, in after_upgrade
    click.style("https://twitter.com/PlatformIO_Org", fg="cyan"))
  File "c:\python27\lib\site-packages\click\utils.py", line 314, in echo
    file.write(message)
  File "c:\python27\lib\site-packages\click\_compat.py", line 560, in _safe_write
    return _write(s)
  File "c:\python27\lib\site-packages\colorama\ansitowin32.py", line 40, in write
    self.__convertor.write(text)
  File "c:\python27\lib\site-packages\colorama\ansitowin32.py", line 141, in write
    self.write_and_convert(text)
  File "c:\python27\lib\site-packages\colorama\ansitowin32.py", line 166, in write_and_convert
    self.write_plain_text(text, cursor, start)
  File "c:\python27\lib\site-packages\colorama\ansitowin32.py", line 174, in write_plain_text
    self.wrapped.write(text[start:end])
  File "c:\python27\lib\site-packages\click\_compat.py", line 97, in write
    return self.buffer.write(str(x))
IOError: [Errno 0] Error

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/en/latest/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio/issues

============================================================

The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

ivankravets, 2017-12-14:
@tekk here is temporary solution http://docs.platformio.org/en/latest/ide/vscode.html#known-issues. You need to do that one time. Let's wait when @Tyriar will fix this issue with VSCode. Otherwise, please switch to Atom where all work without any problems.

tekk, 2017-12-14:
Yup I've already did that. It's a pitty that VS Code integration is so poor compared to Atom. I prefer VS Code... Thanks anyway.

ivankravets, 2017-12-14:
We depend on a few blocked issues by @microsoft for a few months. I think they will fix them next year or in 2020. The process is very slow. I even do not know when this issue with terminal and `IOError: [Errno 0] Error` will be fixed.

tekk, 2017-12-14:
Tried the workaround, with no avail...

```
D:\Code-Git>cd esp-ledstrip-christmas-2017

D:\Code-Git\esp-ledstrip-christmas-2017>pio run -t upload

*************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for IoT development > http://platformio.org/platformio-ide
- donate to keep PlatformIO alive! > http://platformio.org/donate
*************************************************************************************************************************************************************************

Please wait while upgrading PlatformIO ...

Platform teensy
--------
Updating framework-arduinoteensy package:
Versions: Current=9, Latest=9    [Up-to-date]
Updating toolchain-atmelavr package:
Versions: Current=3, Latest=3    [Up-to-date]
Updating ldscripts package:
Versions: Current=23, Latest=23          [Up-to-date]
PlatformIO has been successfully upgraded to 2.11.1!

[12/14/17 19:47:27] Processing lolin32 (platform: espressif32, board: lolin32, framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Unknown platform 'espressif32'
```

ivankravets, 2017-12-14:
> PlatformIO has been successfully upgraded to 2.11.1!

Where did you find it? :) 2015 year. Please remove old versions of PIO Core.

http://docs.platformio.org/en/latest/faq.html#multiple-pio-cores-in-a-system

tekk, 2017-12-14:
Just wanted to take a look at WORKING Atom with PIO to look at what version is it and got this.

```
Uncaught Error: Could not start PIO Home server: Please manually remove file C:\Users\tekks\.platformio\packages\tool-pioplus\pioplus.exe Please manually remove file C:\Users\tekks\.platformio\packages\tool-pioplus\pioplus.exe Error: Could not install 'tool-pioplus' with version requirements '>=0.12.0,<2' for your system 'windows_amd64'. If you use Antivirus, it can block PlatformIO Package Manager. Try to disable it for a while.
C:\Users\tekks\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1571
Show Stack Trace
The error was thrown from the platformio-ide package. You can help by creating an issue. Please explain what actions triggered this error.
Create issue on the platformio-ide package
```
So I rather deleted whole .platformio/packages dir and tried:
```
>pip uninstall platformio
Cannot uninstall requirement platformio, not installed
```
So I deleted whole .platformio directory.
4th time this week :)

I have no idea where the old version is laying.
Any help appreciated, I'm moving away from computer...

ivankravets, 2017-12-14:
@tekk 

1. Close Atom and VSCode.
2. Kill process with `python` and `pioplus`
3. Open OS Control Panel, uninstall ALL Pythons from a system
4. Remove `C:\Python27` folder if exists
5. Remove `%HOME_DIR%/.platformio` folder.
6. Open Atom. Build project. Now, use VSCode.

ivankravets, 2017-12-14:
@Tyriar how can we disable VSCode terminal for tasks? I remember 1.0.0 version of tasks.json has used default Output Panel. Starting from 2.0.0 VSCode switched to a terminal which leads to a ton of problems. I don't see in API how to disable terminal for tasks. Our [source code](https://github.com/platformio/platformio-vscode-ide/blob/develop/src/tasks.js#L246) for Tasks.

- https://code.visualstudio.com/docs/editor/tasks-appendix

-----------

I see **~5,000 IOError** exceptions in our telemetry stats reported by hundred users who use VSCode+Windows for the last days. It seems that we should disable our extension for Windows users to avoid misunderstanding with this issue.

Also, all Python users are affected who use sockets or multithread features.

/cc @dbaeumer

Thanks!

tekk, 2017-12-15:
@ivankravets I did exactly the steps you wrote. (I had 3 different pythons on my system and there was C:\python27 dir still after uninstalling them).

I ran Atom. It installed PlatformIO Core. After restarting, PlafrormIO Home page was stuck (kept loading forever). So after some 5-10 minutes I closed it.
I've tried to build my project, it built it with no problems.
Restarted Atom, to see what version of PIO I currently have (the Home page loaded up OK).
`Home 0.5.3·Core 3.5.0rc8`

Now I closed Atom, and opened VSCode.
PIO Home opened. Loaded in just a few seconds and version was also 
`Home 0.5.3·Core 3.5.0rc8`

Then I tried to compile under VSCode and it also compiled fine.
Excellent. 👍 

Last thing I wanted to try is to update `espressif32` platform from terminal as this was failing before on VSCode.
I opened PlatformIO: New terminal:
```
>pio platform install espressif32
PlatformManager: Installing espressif32
espressif32 @ 0.11.0 is already installed
PackageManager: Installing toolchain-xtensa32 @ ~1.50200.2
Downloading  [------------------------------------]    0%
Error: Please read http://bit.ly/package-manager-ioerror
Warning! Package Mirror: [Errno 0] Error
Looking for other mirror...
Downloading  [------------------------------------]    0%
Error: Please read http://bit.ly/package-manager-ioerror
Warning! Package Mirror: [Errno 0] Error
Looking for other mirror...
Error: Could not install 'toolchain-xtensa32' with version requirements '~1.50200.2' for your system 'windows_amd64'.
If you use Antivirus, it can block PlatformIO Package Manager. Try to disable it for a while.
```

Is this that Known Issue? Even after this the project compiled fine.
I'm happy that I have working VSCode + PIO again!
Thank you very much for your effort and help! 👍 I was already tired of reinstalling over and over again... 

This solved my issues with VSCode and Atom.

dbaeumer, 2017-12-15:
@ivankravets contributed tasks can only be executed in the terminal. There is no support to execute them in the deprecated output runner. Task contribution via extensions is a feature that got introduced with running tasks in the terminal and was never available with task being executed in the output channel.

ivankravets, 2017-12-15:
@dbaeumer Thanks! Now we have a clear picture. I'll add information to our site and warning in our extension that we don't support VSCode for Windows in a case with this bug.

Happy Holidays!

tekk, 2017-12-15:
Please don't abandon VSCode.

Tyriar, 2017-12-18:
FYI the Windows team has fixed this, it will take time to get the update out though https://github.com/Microsoft/vscode/issues/36630

ivankravets, 2017-12-18:
Thanks, @Tyriar  for the info!

Ravenheart, 2018-03-02:
It is now march 2018 and this bug is still not fixed, the PlatformIO IDE comparison chart clearly shows VSCode as the better host IDE and yet something as simple as installing a package basically renders the whole project unusable.

ivankravets, 2018-03-02:
@Ravenheart Thank you very much for the **NEGATIVE** review https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide#review-details. Please read this issue carefully, we can't fix it because the bug is linked with @Microsoft (not with @PlatformIO)

P.S: https://github.com/Microsoft/vscode/issues/36630

Tyriar, 2018-03-03:
@Ravenheart yes there is nothing @ivankravets or even the VS Code team can do about this, it's a regression in Windows which is already fixed but takes a long time to roll out due to the complexity of the project.

ivankravets, 2018-04-06:
Temporary workaround for VSCode issue in https://github.com/platformio/platformio-core/commit/59fe190f205f4175d45457d039b31b6ce5fa2011

ivankravets, 2018-04-06:
Please re-run in terminal `pio upgrade --dev`. Should work now.

closed 2018-04-06 by ivankravets