platformio/platformio-vscode-ide #62: update/upgrade not working

kristsm, 2017-12-15:
Having problems running update/upgrade  commands from VSCODE interface:
Running the contributed command: 'platformio-ide.updateCore' failed.
Running the contributed command: 'platformio-ide.upgradeCore' failed.

Though, it does work running from the terminal.
Windows 10 Pro 1709
VSCODE 1.19.0 (Same errors for 1.18 too) 
PLATFORMIO 3.5.0rc8

ivankravets, 2017-12-16:
Did you open PIO Project?

DJManas, 2017-12-26:
Same problem here.

Mac OS 10.13.2
VSCode 1.19.1 (1.19.1)
PlatformIO: Home 0.5.3·Core 3.5.0rc15

Project opened, not workspace.

kristsm, 2017-12-27:
Yeah, opened project, same as in the screenshot provided in the previous reply

ivankravets, 2017-12-27:
Thanks, fixed! Temporary solution is to use `Menu: Tasks > Run Tasks > Upgrade PIO Core`...

closed 2017-12-27 by ivankravets

DJManas, 2017-12-27:
Thank you.

Regards,
Petr Sourek

commit 6c544ea6 (2017-12-27): Fix issue with PIO Core update/upgrade commands // Resolve #62