platformio/platformio-vscode-ide #63: unable to install platformio extension on ubuntu

zerog2k, 2017-12-19:
Ubuntu 16.04.3 LTS
vscode 1.19.0

When installing platformio ide extension, sort of acts like it's installing, but then says it's successfully installed and to reload. After restart/reload, just keeps in same behavior, and platformio extension never seems to install correctly. If I remove and reinstall, same behavior.

[image]

output from vscode debug:
```
jens@pluto:~$ code --log debug --verbose --inspect-extensions
[main 8:45:16 AM] Starting VS Code
[main 8:45:16 AM] from: /usr/share/code/resources/app
[main 8:45:16 AM] args: { _: [],
  help: false,
  h: false,
  version: false,
  v: false,
  wait: false,
  w: false,
  diff: false,
  d: false,
  add: false,
  a: false,
  goto: false,
  g: false,
  'new-window': false,
  n: false,
  'unity-launch': false,
  'reuse-window': false,
  r: false,
  performance: false,
  p: false,
  'prof-startup': false,
  verbose: true,
  logExtensionHostCommunication: false,
  'disable-extensions': false,
  disableExtensions: false,
  'list-extensions': false,
  'show-versions': false,
  nolazy: false,
  'skip-getting-started': false,
  'sticky-quickopen': false,
  'disable-telemetry': false,
  'disable-updates': false,
  'disable-crash-reporter': false,
  'skip-add-to-recently-opened': false,
  status: false,
  s: false,
  log: 'debug',
  'inspect-extensions': '',
  debugPluginHost: '' }
[main 8:45:16 AM] Resolving machine identifier...
[main 8:45:16 AM] Resolved machine identifier: f7dae92d5c7e6b6cb7b26d653cb9ce400ef4967b5a0b5fb3bc1f807c50445abd
[12720:1219/084517.759814:INFO:CONSOLE(1224)] "%cTRACE", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.main.js (1224)
[main 8:45:18 AM] IPC#vscode-workbenchLoaded
[12720:1219/084518.461163:INFO:CONSOLE(2571)] "%c[Extension Host] %cdebugger listening on port 9333", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.main.js (2571)
[12720:1219/084518.571263:INFO:CONSOLE(4054)] "[/home/jens/.vscode/extensions/ms-vscode.cpptools-0.14.5]: Command `workbench.action.gotoSymbol` appears multiple times in the `commands` section.", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.main.js (4054)
[12720:1219/084518.572265:INFO:CONSOLE(4054)] "[/home/jens/.vscode/extensions/ms-vscode.cpptools-0.14.5]: Command `workbench.action.showAllSymbols` appears multiple times in the `commands` section.", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.main.js (4054)
[12720:1219/084518.765498:INFO:CONSOLE(2569)] "%c[Extension Host] %cdebugger inspector at %cDebugger listening on port 9333.
Warning: This is an experimental feature and could change at any time.
To start debugging, open the following URL in Chrome:
    chrome-devtools://devtools/bundled/inspector.html?experiments=true&v8only=true&ws=127.0.0.1:9333/a8ae547d-8b38-4b16-947f-9af92cc86c21
", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.main.js (2569)
[12720:1219/084519.726398:INFO:CONSOLE(1224)] "%cTRACE", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.main.js (1224)
[12720:1219/084519.740821:INFO:CONSOLE(1224)] "%cTRACE", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.main.js (1224)
[12720:1219/084519.745884:INFO:CONSOLE(249)] "%c[Extension Host] %crunCommand", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.main.js (249)
[12720:1219/084519.760110:INFO:CONSOLE(1225)] "%c  ERR", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.main.js (1225)
[12720:1219/084519.947350:INFO:CONSOLE(249)] "%c[Extension Host] %cTraceback (most recent call last):
  File "/home/jens/.platformio/penv/bin/platformio", line 11, in <module>
    load_entry_point('platformio==3.5.0rc13', 'console_scripts', 'platformio')()
  File "/home/jens/.platformio/penv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 572, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/jens/.platformio/penv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2752, in load_entry_point
    return ep.load()
  File "/home/jens/.platformio/penv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2405, in load
    return self.resolve()
  File "/home/jens/.platformio/penv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2411, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/jens/.platformio/penv/local/lib/python2.7/site-packages/platformio/__main__.py", line 23, in <module>
    from platformio import __version__, exception, maintenance
  File "/home/jens/.platformio/penv/local/lib/python2.7/site-packages/platformio/exception.py", line 230
    "`sudo pio ...`."
    ^
IndentationError: unexpected indent
 %c(at /home/jens/.vscode/extensions/platformio.platformio-ide-0.8.2/lib/installer/manager.js:92:19)", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.main.js (249)
[12720:1219/084520.237292:INFO:CONSOLE(249)] "%c[Extension Host] %crunCommand", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.main.js (249)
[12720:1219/084520.246039:INFO:CONSOLE(249)] "%c[Extension Host] %crunCommand", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.main.js (249)
[12720:1219/084520.255022:INFO:CONSOLE(249)] "%c[Extension Host] %crunCommand", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.main.js (249)
[12720:1219/084520.276601:INFO:CONSOLE(249)] "%c[Extension Host] %crunCommand", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.main.js (249)
[12720:1219/084520.284211:INFO:CONSOLE(249)] "%c[Extension Host] %cUser's Virtualenv: Error: spawn virtualenv ENOENT %c(at PlatformIOCoreStage._callee8$ (/home/jens/.vscode/extensions/platformio.platformio-ide-0.8.2/node_modules/platformio-node-helpers/lib/index.js:2425:25))", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.main.js (249)
[12720:1219/084521.784252:INFO:CONSOLE(249)] "%c[Extension Host] %cactivating extension %c(at Object.activate (/home/jens/.vscode/extensions/ms-vscode.cpptools-0.14.5/out/src/LanguageServer/extension.js:45:13))", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.main.js (249)
[12720:1219/084521.828844:INFO:CONSOLE(249)] "%c[Extension Host] %crunCommand", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.main.js (249)
[12720:1219/084525.332216:INFO:CONSOLE(249)] "%c[Extension Host] %crunCommand", source: file:///usr/share/code/resources/app/out/vs/workbench/workbench.main.js (249)
[12720:1219/084741.215794:WARNING:x11_util.cc(1360)] X error received: serial 4808, error_code 3 (BadWindow (invalid Window parameter)), request_code 3, minor_code 0 (X_GetWindowAttributes)
[12720:1219/084745.449115:WARNING:x11_util.cc(1360)] X error received: serial 5053, error_code 3 (BadWindow (invalid Window parameter)), request_code 3, minor_code 0 (X_GetWindowAttributes)

```

zerog2k, 2017-12-19:
looks like there is some problem here?
```
[12720:1219/084519.947350:INFO:CONSOLE(249)] "%c[Extension Host] %cTraceback (most recent call last):
  File "/home/jens/.platformio/penv/bin/platformio", line 11, in <module>
    load_entry_point('platformio==3.5.0rc13', 'console_scripts', 'platformio')()
  File "/home/jens/.platformio/penv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 572, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/jens/.platformio/penv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2752, in load_entry_point
    return ep.load()
  File "/home/jens/.platformio/penv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2405, in load
    return self.resolve()
  File "/home/jens/.platformio/penv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2411, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/jens/.platformio/penv/local/lib/python2.7/site-packages/platformio/__main__.py", line 23, in <module>
    from platformio import __version__, exception, maintenance
  File "/home/jens/.platformio/penv/local/lib/python2.7/site-packages/platformio/exception.py", line 230
    "`sudo pio ...`."
    ^
IndentationError: unexpected indent
```

arendst, 2017-12-19:
Same on windows 10

zerog2k, 2017-12-19:
looks like due to https://github.com/platformio/platformio-core/issues/1251
I opened the extension.py and manually appended a backslash, to line 229, and now it seems to go further..
(Not sure how that slipped passed any CI testing ....)

arendst, 2017-12-19:
Reloaded many times and seems to work now too.

ivankravets, 2017-12-23:
We fixed it, it was typo. Sorry :(

closed 2017-12-23 by ivankravets