platformio/platformio-vscode-ide #65: Project Wizard "choose a directory where we will create project file " is Loading forever

DCMay, 2017-12-25:
Am using vsc(Update 1.19.1),and I installed the **official**  platform ide from extention,but int Project Wizard "choose a directory where we will create project file " is Loading forever like this:
[image]
what's going on?

ivankravets, 2017-12-27:
What is your OS? Please open PIO IDE Terminal and type `pio home`. Do you see any errors?

DCMay, 2017-12-27:
@ivankravets  hi, thanks for replay. am using windows 10. and I did something like this: 
[image]
it all good, it opened a new tab of platformio home page in my chrome.
but the problem remain. Choose a directory for create project files it's loading forever......

ivankravets, 2017-12-30:
Please type `pio device list --logical`. What do you see now?

DCMay, 2018-01-08:
@ivankravets  

[image]

ivankravets, 2018-01-08:
Thanks! It seems that a bug and moved to PIO Core!

https://github.com/platformio/platformio-core/issues/1286

closed 2018-01-08 by ivankravets