platformio/platformio-vscode-ide #66: Only activate extension when intending to use PlatformIO

WasabiFan, 2018-01-01:
Currently, the extension always activates and in many ways "takes over" the editor whenever you load it, even if you aren't working on a PlatformIO project: attempts to check for updates, adds status bar items, creates files in your working directory (!!!), etc.

Can the behavior be changed to only activate the extension when it's needed? Currently, `package.json` defines `activationEvents` to be `*`, meaning "always active". The more surgical approach would be to activate when a command is run and/or when [platformio.ini is present](https://code.visualstudio.com/docs/extensionAPI/activation-events#_activationeventsworkspacecontains).

This would partially help #64 as well, although the better solution for that would be to not dump the whole PATH into an otherwise SCM-able settings file.

wsnell, 2018-01-13:
I agree, only if the ini is present. Can you have a project file that is opened?

eosrei, 2018-01-16:
Littering `.vscode/settings.json` changes in every project directory needs to stop. I use VScode for the rest of my development work. I can run Atom instead in the meantime though. Thanks!

WasabiFan, 2018-01-16:
@eosrei My solution has been to entirely disable the extension, and then use the "Enable (workspace)" option to only load it in the single project I use this extension in. That should be a bit easier for users to do than editing the manifest, but it doesn't fix #64 and certainly doesn't help the majority of users who suddenly have their project files messed up without understanding the cause.

@ivankravets _please_ take a look at this!

ivankravets, 2018-01-16:
@eosrei 
> Littering .vscode/settings.json changes in every project directory needs to stop. 

Did you disable [updateTerminalPathConfiguration]( http://docs.platformio.org/en/latest/ide/vscode.html#platformio-ide-updateterminalpathconfiguration)?

@WasabiFan 

There is a big problem here. We can't depend on `platformio.ini` because new users will have a problem to create a new project. So, if you don't have a PlatformIO-based project, you will not see PIO Home, etc.
I think the single solution here is to create a new setting where each of you will be able to decide the behaviour of this extension. For example, `activateOnlyForPlatformIOProject` which will be set to `false` by default.

WasabiFan, 2018-01-16:
Here's my suggestion — technical note at the bottom.

1. On first run of VSCode following extension installation, open an info message which tells the user the command to create a PlatfotmIO project. Perhaps with a button to open a help page.
2. On all loads, only register commands initially; do not open Home, add any status bar entries, etc. _The key is to be as lightweight as possible._ Do a single check to see if `platformio.ini` is present in the workspace.
3. If the user runs the PlatformIO "create" command or if the config file was discovered in the workspace, load the full extension. One command available is "deactivate" which reverts to the inactive state to unclutter the UI.

One of the big advantages of this design is that there are no settings you must discover. The extension intelligently activates when you need it and gets out of the way otherwise, for all users.

On the technical side, you'll need to retain `activationEvents: *` as there is no event for "first run". As long as you do absolutely nothing but register commands and a single file check, any perf penalty will be negligible.

ivankravets, 2018-01-16:
@WasabiFan that was the initial workflow for the first versions of this extension. We were attacked by a community with a dozen of questions how to use it. This is very good that you are familiar with VSCode and use its commands, etc. 99% of our customers see VSCode first time.

We have 80K downloads for the last 6 months and only 4 developers participate in this issue. It means that current workflow works well for 99% our customers.

But! Of course, we respect our community and hear your feedback. Please update to 0.11.0 and enable `activateOnlyOnPlatformIOProject`  option.

- https://github.com/platformio/platformio-vscode-ide/releases/tag/v0.11.0

eosrei, 2018-01-16:
> Did you disable updateTerminalPathConfiguration?

No, because I didn't know it was optional. Changing adding files automatically is a bug... so to the issue queue!

> We have 80K downloads for the last 6 months and only 4 developers participate in this issue.

I've used the same argument myself in my projects ;) Although to be fair, this issue hasn't been open for 6 months. :laughing: 

This resolution works for me.

MickPBduece, 2019-03-14:
As a newbie - this seems like the most rediculous way to act as a group I've seen in a while.  I will certainly not think of contributing to PlatformIO and that hideous orange bug I am trying to suppress.

ivankravets, 2019-03-14:
@MickPBduece this issue is fixed. Do you still experience any problems?

MickPBduece, 2019-03-17:
I did until I added a settings.jsom entry to only PIO for specific
workspaces

Thanks.

clplaneguy, 2019-03-18:
I to have an extension conflict with Arduino.  What is the settings.json entry?

MickPBduece, 2019-03-18:
add to settings.json
"platformio-ide.activateOnlyOnPlatformIOProject": true,

ivankravets, 2019-03-20:
We have a lot of other settings. See http://docs.platformio.org/en/latest/ide/vscode.html#settings

TurnOffNOD, 2022-04-02:
> 

Hi,, after I have set "activateOnlyOnPlatformIOProject" to true, how can I bring up platformio home?

ivankravets, 2022-04-02:
@TurnOffNOD open any PlatformIO project (a folder that contains `platformio.ini` file)

TurnOffNOD, 2022-04-02:
> @TurnOffNOD open any PlatformIO project (a folder that contains `platformio.ini` file)

Oh. But what if I want to create a new PlatformIO project without a existing one?

novirium, 2022-04-13:
It's a bit awkward, but the way I've been doing this is to just open a blank folder _somewhere_ with a blank `platformio.ini` file in it. That then triggers the extension, and I can use the New Project wizard as normal.

closed 2022-12-11 by ivankravets

ivankravets, 2022-12-11:
Fixed in the latest development version (upcoming PlatformIO IDE 3.0). Please re-test with the pre-release version. See instructions below:

[image]

commit d4cc024c (2018-01-16): Added new option `activateOnlyOnPlatformIOProject` // Resolve #66
commit f66ef944 (2022-12-11): Activate PlatformIO IDE extension when intending to use PlatformIO // Resolve #66