platformio/platformio-vscode-ide #68: VSCode: PlatformIO IntelliSense Index Rebuild task runs forever

learmond, 2018-01-09:
### Configuration

**Operating system**:
Windows 10 Pro 64-bit v1709

**PlatformIO Version** (`platformio --version`):
Home: 0.6.0, Core: 3.5.1a3

### Description of problem
Upon creating a new project with the same board/configuration as I have done before, the 'IntelliSense Index Rebuild' task which automatically runs on startup will not stop, and this prevents building/deploying of the project code.

All other projects created before this point did not suffer from this effect. A previous project using the same board and platformIO.ini worked perfectly. Once this new project was created (in the same way I've done in the past), all projects now suffer from this infinite IntelliSense Index Rebuild.

Restarting all instances of VSCode did not help and restarting the computer did not help.

Triggering a build while in this state only outputs the build info:
```
> Executing task: platformio.exe run <

.. warnings .. (not applicable to this problem)
[.. timestamp ..] Process featheresp32 (<platformio.ini config>... same as when it worked>)
------------------------------------------------------------------------------------------------------------
```
... then nothing. Just the spinner in the bottom toolbar next to "PlatformIO: IntelliSense Index Rebuild"

Does anyone have any ideas?
I'm thinking my next move is to re-install PlatformIO extension.

learmond, 2018-01-10:
Re-installed PlatformIO ----------------------------------  didn't help.
Removed 32 bit VSCode, installed 64-bit version ------- didn't help.
Opened up my project folder without internet access (disabled adapter) ----- Success ?!?!
- IntelliSense completes in a couple seconds, and I can successfully build and deploy to the board.

If I enable my adapter again, then manual trigger the IntelliSense re-build task, the problem returns... the task doesn't complete.

Also of note, the Libraries section of PIO Home does not complete its loading routine. The loading spinner just continues forever. This happens regardless of whether the IntelliSense task is currently stuck running, or whether it has successfully completed (and after re-enabling the network adapter).

ivankravets, 2018-01-10:
Could you try to run `pio run` manually? Also, do you use the latest VSCode?

learmond, 2018-01-10:
Hey! Thanks for the assistance.
I've installed the latest available 64-bit VSCode as of yesterday. VSCode - About details:
```
Version 1.19.1
Commit 0759f77bb8d86658bc935a10a64f6182c5a1eeba
Date 2017-12-19T09:46:23.884Z
Shell 1.7.9
Renderer 58.0.3029.110
Node 7.9.0
Architecture x64
```

Executing `pio run` manually in the VSCode terminal results in the same behavior, specifically, when internet is available, it doesn't make it past the first bit as outlined in my first post, then when my ethernet adapter is disabled, it successfully completes the build.

NOTE: POI Home still cannot successfully load the Registry's libraries.

ivankravets, 2018-01-10:
Try to remove `%HOME_DIR%/.platformio` folder and start building project. You will need Internet

learmond, 2018-01-10:
For the following, internet is connected...
I shut down VSCode and removed `.platformio/` from my user's home directory.
Upon starting VSCode, which automatically opened my project directory, it automatically installed the PlatformIO extension which completed successfully.
After reloading the window, PIO Home tried to automatically open, but could not and eventually failed due to timeout. The following is the dev tools console output on reload:

[image]

Let me know if more detailed stack traces would help.

I thinking this may be due to my custom library directories setup. It was working before creating this last project, so I didn't think it was an issue, however it would be good to see if I'm doing this correctly, or if there is a better way of accomplishing my goals.

I'm using the adafruit esp32 feather board with gps and LoRa peripherals. For libraries, I'm using the Tiny GPS (https://github.com/mikalhart/TinyGPS.git), and LoRa (https://github.com/sandeepmistry/arduino-LoRa).
These were originally installed via PIOHome's Library Registry into a custom Storage location made available in the UI by specifying it in my project's platformio.ini as follows:
```
[platformio]
lib_extra_dirs = </path/to/custom/libs_dir>
```
This allowed it to show up in the Storage list in the "Install To..." section.

The libs in this custom folder are shared between multiple esp32 projects and are revisioned in my own VCS (not necessarily important,). I needed to make changes to the LoRa library, so I branched from the custom storage location to another location which I then added to the platformio.ini file:
```
[platformio]
lib_extra_dirs = </path/to/custom/libs_dir>, </path/to/branched_libs>
```
I renamed the branched library in its `.library.json` file (ie. 'custLoRa'), and after much trial and error, got it to override the original by turning off LDF and manually specifying which libraries I want in my specific environment:
```
[env:featheresp32]
platform = espressif32
board = featheresp32
framework = arduino
lib_ldf_mode = off
lib_deps = SPI, custLoRa, TinyGPS
```
I also had shared headers in the branched libs directory, so I added it as an include location:
```
[platformio]
lib_extra_dirs = </path/to/custom/libs_dir>, </path/to/branched_libs>
include_dir = </path/to/branched_libs>
```

NOTE: `</path/to/branched_libs>` is actually a relative path and exists in the project's parent directory.

All I really want to do is customize a library and have it override the original in a given project. Ideally, i'd like to have LDF set to default, so if there is a better way of doing this, I'd love to hear it.

Thanks again.

ivankravets, 2018-01-10:
Could you try this solution https://github.com/platformio/platformio-atom-ide/issues/1298#issuecomment-353728372 ?

ivankravets, 2018-01-23:
Please reopen if you still need help

closed 2018-01-23 by ivankravets

alexgalsyan, 2018-05-22:
Hello I have a same problem, and need help.

orvisevans, 2019-12-31:
Hi there,
I had the same problem (Just the spinner in the bottom toolbar next to "PlatformIO: IntelliSense Index Rebuild") and solved it the same way (disabling the wifi adaptor on my laptop).

Should we reopen this issue, should I start a new one, or is this considered not an issue?