platformio/platformio-vscode-ide #69: The system cannot find the path specified when build task is run

nickdex, 2018-01-12:
Since I have updated platformio extension, when build/upload I'm getting this error.
Here's link to my repo - [Pubnub-Led](https://github.com/nickdex/Rgb-Bulb-Arduino/tree/develop)
## Terminal Output

```
> Executing task: platformio.exe run <

[01/12/18 09:46:09] Processing nodemcuv2 (platform: espressif8266; lib_deps: Pubnub, ArduinoJson, ESP8266WiFi; board: nodemcuv2; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Converting pubnub-led.ino
The system cannot find the path specified.
*** Error 1
Collected 33 compatible libraries
Scanning dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <Pubnub> v2.1.1
|-- <ArduinoJson> v5.11.1
|-- <ESP8266WiFi> v1.0
|-- <RGBColors>
Compiling .pioenvs\nodemcuv2\src\pubnub-led.ino.o
Archiving .pioenvs\nodemcuv2\libFrameworkArduinoVariant.a
The system cannot find the path specified.
The system cannot find the path specified.
Compiling .pioenvs\nodemcuv2\FrameworkArduino\Esp.o
Compiling .pioenvs\nodemcuv2\FrameworkArduino\FS.o
The system cannot find the path specified.
The system cannot find the path specified.
*** [.pioenvs\nodemcuv2\libFrameworkArduinoVariant.a] Error 1
*** [.pioenvs\nodemcuv2\FrameworkArduino\Esp.o] Error 1
*** [.pioenvs\nodemcuv2\src\pubnub-led.ino.o] Error 1
*** [.pioenvs\nodemcuv2\FrameworkArduino\FS.o] Error 1
==================================================================================== [ERROR] Took 5.46 seconds ====================================================================================
The terminal process terminated with exit code: 1
```

ivankravets, 2018-01-12:
Try to remove `.pioenvs` folder and start building again.

nickdex, 2018-01-13:
Getting the same error even after removing `.pioenvs` directory

ivankravets, 2018-01-13:
Could you open PIO IDE Terminal (button in toolbar) and type `pio run -v`. Do you see these errors?

nickdex, 2018-01-13:
Yes, I see the same errors. Here's the output -
```
[01/13/18 21:34:26] Processing nodemcuv2 (platform: espressif8266; lib_deps: Pubnub, ArduinoJson, ESP8266WiFi; board: nodemcuv2; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Storage: C:\Users\nikwa\Documents\PlatformIO\Projects\Pubnub-Led\.piolibdeps
LibraryManager: Installing id=1183
Pubnub @ 2.1.1 is already installed
LibraryManager: Installing id=64
ArduinoJson @ 5.11.1 is already installed
Looking for ESP8266WiFi library in registry
xtensa-lx106-elf-g++ -o "C:\Users\nikwa\Documents\PlatformIO\Projects\Pubnub-Led\src\pubnub-led.ino.cpp" -x c++ -fpreprocessed -dD -E "c:\users\nikwa\appdata\local\temp\tmpumclba"
The system cannot find the path specified.
*** Error 1
Collected 33 compatible libraries
Scanning dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <Pubnub> v2.1.1 (C:\Users\nikwa\Documents\PlatformIO\Projects\Pubnub-Led\.piolibdeps\Pubnub_ID1183)
|-- <ArduinoJson> v5.11.1 (C:\Users\nikwa\Documents\PlatformIO\Projects\Pubnub-Led\.piolibdeps\ArduinoJson_ID64)
|-- <ESP8266WiFi> v1.0 (C:\Users\nikwa\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi)
|-- <RGBColors> (C:\Users\nikwa\Documents\PlatformIO\Projects\Pubnub-Led\lib\RGBColors)
xtensa-lx106-elf-ar rc .pioenvs\nodemcuv2\libFrameworkArduinoVariant.a
xtensa-lx106-elf-g++ -o .pioenvs\nodemcuv2\src\pubnub-led.ino.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -ffunction-sections -fdata-sections -DARDUINO=10600 -DLWIP_OPEN_SRC -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=30501 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -U__STRICT_ANSI__ -IC:\Users\nikwa\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include -IC:\Users\nikwa\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip\include -IC:\Users\nikwa\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include -IC:\Users\nikwa\.platformio\packages\framework-arduinoespressif8266\cores\esp8266 -IC:\Users\nikwa\.platformio\packages\framework-arduinoespressif8266\variants\nodemcu -Isrc -I.piolibdeps\Pubnub_ID1183 -I.piolibdeps\ArduinoJson_ID64\src -IC:\Users\nikwa\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src -Ilib\RGBColors src\pubnub-led.ino.cpp
The system cannot find the path specified.
The system cannot find the path specified.
xtensa-lx106-elf-g++ -o .pioenvs\nodemcuv2\FrameworkArduino\Esp.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -ffunction-sections -fdata-sections -DARDUINO=10600 -DLWIP_OPEN_SRC -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=30501 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -U__STRICT_ANSI__ -IC:\Users\nikwa\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include -IC:\Users\nikwa\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip\include -IC:\Users\nikwa\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include -IC:\Users\nikwa\.platformio\packages\framework-arduinoespressif8266\cores\esp8266 -IC:\Users\nikwa\.platformio\packages\framework-arduinoespressif8266\variants\nodemcu C:\Users\nikwa\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\Esp.cpp
xtensa-lx106-elf-g++ -o .pioenvs\nodemcuv2\FrameworkArduino\FS.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -ffunction-sections -fdata-sections -DARDUINO=10600 -DLWIP_OPEN_SRC -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=30501 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -U__STRICT_ANSI__ -IC:\Users\nikwa\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include -IC:\Users\nikwa\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip\include -IC:\Users\nikwa\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include -IC:\Users\nikwa\.platformio\packages\framework-arduinoespressif8266\cores\esp8266 -IC:\Users\nikwa\.platformio\packages\framework-arduinoespressif8266\variants\nodemcu C:\Users\nikwa\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\FS.cpp
The system cannot find the path specified.
The system cannot find the path specified.
*** [.pioenvs\nodemcuv2\libFrameworkArduinoVariant.a] Error 1
*** [.pioenvs\nodemcuv2\FrameworkArduino\FS.o] Error 1
*** [.pioenvs\nodemcuv2\FrameworkArduino\Esp.o] Error 1
*** [.pioenvs\nodemcuv2\src\pubnub-led.ino.o] Error 1
```

ivankravets, 2018-01-13:
I tried your project in VM and can't reproduce.

Please try this solution for a test http://docs.platformio.org/en/latest/faq.html#solution-3-run-from-terminal

So, please close IDE, open system terminal, then run PIO Core directly in terminal. Does it work?

nickdex, 2018-01-13:
Followed all the steps as instructed, again getting same errors. There was one strange thing that happened that I forgot to mention. I had platformio for my angular project workspace. I opened this project after couple of months and it had asked me to again install platformio ide (although it worked in other workspaces for eg got toolbar and opening the pio home etc). I installed it and reloaded vscode and started getting these errors

wsnell-qualcomm, 2018-01-16:
try witin code run the platformio terminal and then type pio upgrade.  That fixed issues for me.

nickdex, 2018-01-17:
Here's what I get. I think the this problem started when upgraded PIO
```
You're up-to-date!
PlatformIO 3.5.1a7 is currently the newest version available.
```

nickdex, 2018-01-21:
PlatformIO updated to 3.5.1, still getting that error, this is also replicated in atom's ide installation. However when I installed it in my macbook, its works fine. 
Can you suggest a way to purge all platform ide files from my system? (Windows 10)
I think this issue will get fixed with a fresh install

ivankravets, 2018-02-02:
@nickdex have you resolved this issue?

nickdex, 2018-02-02:
No, I removed everything related to platform io with a file sniffer and reinstalled it but still getting that error.

shadowkrazee, 2018-05-16:
I seem to be having this same issue, and  have tried everything listed in this thread including:
> -uninstall / reinstall PlatformIO AND atom
-tried several other projects, including the "Blink" arduino example
-running pio from a system cmd prompt
-did a pio upgrade, upgrade success, same errors after
-deleted .pioenvs directory and tried rebuilding

all to no avail.
Running Windows 10, 64 bit hardware

when doing "pio run -v" on my project in a terminal (inside atom, or a cmd prompt) i get this:
`c:\Users\johnathan\Documents\PlatformIO\Projects\180514-211256-nanoatmega328>pio run -v                                                                                                                 [05/15/18 22:46:20] Processing nanoatmega328 (platform: atmelavr; board: nanoatmega328; framework: arduino)                                                                                             ----------------------------------------------------------------------------------------------------------------------------                                                                            PLATFORM: Atmel AVR > Arduino Nano ATmega328                                                                                                                                                            SYSTEM: ATMEGA328P 16MHz 2KB RAM (30KB Flash)                                                                                                                                                           Library Dependency Finder -> http://bit.ly/configure-pio-ldf                                                                                                                                            LDF MODES: FINDER(chain) COMPATIBILITY(light)                                                                                                                                                           Framework incompatible library C:\Users\johnathan\.platformio\lib\SoftwareSerial_ID2728                                                                                                                 Collected 25 compatible librariesMore details about "Library Compatibility Mode": http://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode                                                                                                                                                                                                                                                       Scanning dependencies...                                                                                                                                                                                Dependency Graph                                                                                                                                                                                        |-- <FastLED> v3.1.8 (C:\Users\johnathan\.platformio\lib\FastLED_ID126)                                                                                                                                 |   |-- <SoftwareSerial> v1.0 (C:\Users\johnathan\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SoftwareSerial)                                                                 avr-g++ -o .pioenvs\nanoatmega328\liba56\FastLED_ID126\FastLED.cpp.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=30503 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -DARDUINO=10805 -IC:\Users\johnathan\.platformio\packages\framework-arduinoavr\cores\arduino -IC:\Users\johnathan\.platformio\packages\framework-arduinoavr\variants\eightanaloginputs -Isrc -IC:\Users\johnathan\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SoftwareSerial\src -IC:\Users\johnathan\.platformio\lib\FastLED_ID126 C:\Users\johnathan\.platformio\lib\FastLED_ID126\FastLED.cpp                                                                                                            The system cannot find the path specified.                                                                                                                                                              avr-g++ -o .pioenvs\nanoatmega328\liba56\FastLED_ID126\bitswap.cpp.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=30503 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -DARDUINO=10805 -IC:\Users\johnathan\.platformio\packages\framework-arduinoavr\cores\arduino -IC:\Users\johnathan\.platformio\packages\framework-arduinoavr\variants\eightanaloginputs -Isrc -IC:\Users\johnathan\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SoftwareSerial\src -IC:\Users\johnathan\.platformio\lib\FastLED_ID126 C:\Users\johnathan\.platformio\lib\FastLED_ID126\bitswap.cpp                                                                                                            The system cannot find the path specified.                                                                                                                                                              avr-g++ -o .pioenvs\nanoatmega328\liba56\FastLED_ID126\colorpalettes.cpp.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -Os -Wall -ffunction-sections -fdata-sections -flto  [... cut ...]      *** [.pioenvs\nanoatmega328\liba56\FastLED_ID126\colorutils.cpp.o] Error 1                                                                                                                              ================================================ [ERROR] Took 1.52 seconds ================================================`
(Sorry, i couldn't get the lines to unwrap like the other posts..)
the funny thing is, i can browse to these files in explorer, and THEY DO EXIST
Any ideas from you guys?

shadowkrazee, 2018-05-16:
In addition to my last post, I installed atom and platformio in ubuntu (on the same PC), and tried to build the same project...
IT WORKED LIKE A CHARM!
output below:

`[Wed May 16 16:41:37 2018] Processing nanoatmega328 (platform: atmelavr; board: nanoatmega328; framework: arduino)
--------------------------------------------------------------------------------
PLATFORM: Atmel AVR > Arduino Nano ATmega328
SYSTEM: ATMEGA328P 16MHz 2KB RAM (30KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 25 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <FastLED> v3.1.8 (/home/johnathan/.platformio/lib/FastLED_ID126)
|   |-- <SoftwareSerial> v1.0 (/home/johnathan/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/SoftwareSerial)
avr-size --mcu=atmega328p -C -d .pioenvs/nanoatmega328/firmware.elf
AVR Memory Usage
----------------
Device: atmega328p

Program:    4664 bytes (14.2% Full)
(.text + .data + .bootloader)

Data:        323 bytes (15.8% Full)
(.data + .bss + .noinit)

========================= [SUCCESS] Took 0.60 seconds =========================
`
I hope this helps you guys find out what's going on..

shadowkrazee, 2018-05-16:
Update to previous post:
I installed atom and PlatformIO on ubuntu (running on the same PC) and attempted to build thew same project..
AND IT WORKED! output below:
`[Wed May 16 16:41:37 2018] Processing nanoatmega328 (platform: atmelavr; board: nanoatmega328; framework: arduino)
PLATFORM: Atmel AVR > Arduino Nano ATmega328
SYSTEM: ATMEGA328P 16MHz 2KB RAM (30KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 25 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <FastLED> v3.1.8 (/home/johnathan/.platformio/lib/FastLED_ID126)
|   |-- <SoftwareSerial> v1.0 (/home/johnathan/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/SoftwareSerial)
avr-size --mcu=atmega328p -C -d .pioenvs/nanoatmega328/firmware.elf
AVR Memory Usage
Device: atmega328p

Program:    4664 bytes (14.2% Full)
(.text + .data + .bootloader)

Data:        323 bytes (15.8% Full)
(.data + .bss + .noinit)

========================= [SUCCESS] Took 0.60 seconds =========================`

Hope this helps you guys figure it out..

ivankravets, 2018-05-16:
Do you have Antivirus software in a system? Try to disable it from for a while. It seems that it blocks source code builder treating it as an virus which produces object files.

shadowkrazee, 2018-05-16:
I only use windows defender in windows 10..
could i add an exception for a particular directory or process?
if so, what would that be?
Thanks for the quick response!

nickdex, 2018-06-03:
I too have only windows defender in Windows 10 and following same steps as @shadowkrazee didn't make any difference

adryyy, 2019-03-14:
Fixed it by deleting AutoRun key from registry and also remove invalid paths from environment variables.
https://www.tenforums.com/general-support/122582-cmd-terminal-starts-system-cannot-find-path-specified.html

nickdex, 2019-03-16:
I switched to Arch Linux while back so can't test it. @shadowkrazee If you are still using W10; can you  try this solution?

punisher97, 2019-10-30:
> Se corrigió eliminando la clave AutoRun del registro y también elimina las rutas no válidas de las variables de entorno.
> https://www.tenforums.com/general-support/122582-cmd-terminal-starts-system-cannot-find-path-specified.html
This solves the problem, it seems that it is generated by anancoda, I saw that anancoda and platformio do not get along, thanks!

I hope someone else serves you

I had a previous installation of anancoda that prevented installing platformio, I uninstalled it but nevertheless I could not compile since it has a hook in the console of commands that are executed when compiling platform
[image]

ali80, 2020-01-27:
> 
> 
> Fixed it by deleting AutoRun key from registry and also remove invalid paths from environment variables.
> https://www.tenforums.com/general-support/122582-cmd-terminal-starts-system-cannot-find-path-specified.html

this fixed the issue for me but why?

ivankravets, 2020-01-28:
Wow! I'm really shocked. What is inside of this hook.bat? 🐛 ? 🐞 ? 👎

ali80, 2020-01-28:
> 
> 
> Wow! I'm really shocked. What is inside of this `conda_hook.bat`? 🐛 ? 🐞 ? 👎
> 
> Please uninstall conda from a system and use the latest official Python 3. See instructions here https://docs.platformio.org/en/latest/faq.html#install-python-interpreter

Well I need both Anaconda and PIO. This is some annoying stuff specially if you don't know the solution, anyway after applying the registry fix they both are working alongside each other

ivankravets, 2020-01-28:
How can we help when you don’t want to provide information? I don’t use Anaconda and don’t want install it. Could you provide contents of that BAT file?

ali80, 2020-01-28:
I didn't get that you asked for it, 
```@REM Copyright (C) 2012 Anaconda, Inc
@REM SPDX-License-Identifier: BSD-3-Clause
@REM The file name is conda_hook.bat rather than conda-hook.bat because conda will see
@REM the latter as a 'conda hook' command.

@IF DEFINED CONDA_SHLVL GOTO :EOF

@FOR %%F in ("%~dp0") do @SET "__condabin_dir=%%~dpF"
@SET "__condabin_dir=%__condabin_dir:~0,-1%"
@SET "PATH=%__condabin_dir%;%PATH%"
@SET "CONDA_BAT=%__condabin_dir%\conda.bat"
@FOR %%F in ("%__condabin_dir%") do @SET "__conda_root=%%~dpF"
@SET "CONDA_EXE=%__conda_root%Scripts\conda.exe"
@SET __condabin_dir=
@SET __conda_root=

@DOSKEY conda="%CONDA_BAT%" $*

@SET CONDA_SHLVL=0
```
this is for conda version 4.7.12 which I believe is the last version at the moment

ivankravets, 2020-01-28:
This issue contains a lot of comments. The `conda_hook.bat` looks OK for me, they just patch system environment varaible. So, which problems does this script make for you?

ali80, 2020-01-28:
So PIO project compiled without problem and after installing Anaconda I got this error for a lot of .o files 
`The system cannot find the path specified.` as mentioned by the first post.
cleaning and also deleting `.pioenvs` dir also didnt help

ivankravets, 2020-01-28:
They override the system environment PATH and remove other PATHs that were installed in the session. You can remove this conda startup script and just add it to your system path manually.

closed 2020-09-18 by stale[bot]

DotNaos, 2023-04-11:
THANK YOU SOOOOO much, I also had this problem and it was also the problem I had with my Stable Diffusion installation

nve-sak, 2024-05-20:
Thanks, @msvargas, that registry Anaconda cleanup fixed the problem for me as well :-)