platformio/platformio-vscode-ide #70: Error : Installation

edwin6938, 2018-01-12:
VSCode Version: 1.19.1
OS Version: Windows 7 Enterprise
Steps to Reproduce:

I tried to install PlatformIO Core
After few second I am asked if python 2.7 is already install. If so, enter the path... That's what I do
After few second, I receive a msgbox error (cf. enclosed)

[image]

Thank you for your help

ivankravets, 2018-01-12:
Do you use a proxy? Something is wrong with an Internet connection.

wsnell-qualcomm, 2018-01-16:
I had multiple problems with installing the PlatformIO extension and found them all to be related to python.  So I finally unloaded Python 3.5.x and 2.7.x and then reinstalled 2.7.14 with the option to enable path and then everything worked fine.  I would try this.

ivankravets, 2018-01-16:
Do you have Cygwin in a system?

closed 2018-02-02 by ivankravets