platformio/platformio-vscode-ide #71: Change launch.json file to use relative pathames vs absolute path names to allow easier usage on different machines or by different userers with different paths

wsnell, 2018-01-13:
In file launch.json (you should change the values of "target" and "gdbpath", to remove the hardcoded absolute paths that will change for each user grabbing your project from GIT or other, and instead replace with the substitute path keys that VSStudio provides so other users won't have a conflict). See update launch.json file to see how it should be.
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "gdb",
            "request": "launch",
            "cwd": "${workspaceRoot}",
            "name": "PlatformIO Debugger",
            "target": "${workspaceFolder}/.pioenvs/lpc1768/firmware.elf",
            "gdbpath": "${env:USERPROFILE}/.platformio/penv/Scripts/piodebuggdb",
            "autorun": [ "source .pioinit" ]
        }
    ]
}

ivankravets, 2018-01-18:
This file is excluded from VCS. No need to share it in public repositories because PIO Core will re-create automatically working `launch.json`.

See https://github.com/platformio/platformio-core/blob/develop/platformio/ide/tpls/vscode/.gitignore.tpl#L4

ivankravets, 2018-02-02:
As described in the comment above, there is no sense to place this file to VCS. It will be overwritten on the first IDE start

closed 2018-02-02 by ivankravets

nospam2000, 2019-01-05:
> As described in the comment above, there is no sense to place this file to VCS. It will be overwritten on the first IDE start

Yes and this is a really bad idea! Today I had to restore it 50 times because it contains custom configuration values. I start to dislike PlatformIO and it's automatic mechanisms (which can not always be right) and switch back to Eclipse.

It seems you have never used more complicated debugging setups like this one:
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "gdb",
            "request": "launch",
            "name": "OpenOCD Launch gdb",
            "target": "${workspaceRoot}/.pioenvs/lolin_d32_pro:debug/firmware.elf",
            "cwd": "${workspaceRoot}",
            "gdbpath": "/Users/michael/.platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-gdb",
            "printCalls": true, "showDevDebugOutput": true,
            "autorun": [
                "set remotetimeout 60",
                "set remote hardware-watchpoint-limit 2",
                "target remote | /Users/michael/esp/openocd-esp32/bin/openocd -c \"gdb_port pipe\" --search /Users/michael/esp/openocd-esp32/share/openocd/scripts -f interface/jlink.cfg -f board/esp32-wrover.cfg -c \"debug_level 2;init\"",
                "mon reset halt",
                "flushregs",
                "thb setup",
                "c",
            ],
        }
}
```
 Michael

AlexFBP, 2019-01-09:
What a shame that a so nice IDE for microcontrollers, has to be so envious (forcing PioPlus) with a single file that should not be overwriten as it's a VSCODE file... (in the .vscode folder...)

https://community.platformio.org/t/vscode-ide-automatically-generate-of-launch-json-every-time/3305/3

AlexFBP, 2019-01-09:
However, a workaround you can use is to apply the same technique, through [Advanced Scripting](https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html) in your platformio.ini, to make it launch a Python script that deletes the created launch.json and replaces it by other, like in https://github.com/AsharaStudios/GuruxDLMS.c/commit/3eb312042ba4d82b798db7afd39e8219f49295ab

nospam2000, 2019-01-09:
another workaround is to make the file read-only (`chmod a-w .vscode/launch.json`) when you no longer change it, but the you have to remove the protection when you want to edit it

ivankravets, 2019-01-09:
1. You can turn-off index rebuilding with https://docs.platformio.org/en/latest/ide/vscode.html#platformio-ide-autorebuildautocompleteindex

2. "more complicated debugging setups", you can do the same with `platformio.ini`. The settings will be cross-compatible between OSes. See https://docs.platformio.org/en/latest/projectconf/section_env_debug.html

3. "forcing PioPlus". Stay tuned with us on https://twitter.com/PlatformIO_Org. We are working on this.

Happy New Year!

AlexFBP, 2019-01-09:
KISS principle applied hahaha :+1: @nospam2000 However, I'm not so sure if in Windows platforms this is going to work, I'm sure that if I commit the file after `chmod` it's going to work when cloned/pulled from Linux

AlexFBP, 2019-01-09:
The thing @ivankravets Is, I have been reading that documentation section_env_debug, and also the ones placed at the beginning of the generated `launch.json`, but I don't see a clear example that shows  how specific `platformio.ini` directive(s) gets translated in a custom config in `launch.json`

Possibly I'm a dumb and may I be omiting something...