platformio/platformio-vscode-ide #72: pyocd and debugging

wsnell, 2018-01-13:
Can you do a better job with auto configuring for debugger support on vscode. Took me forever to figure out that I had to both install pyocd via pip and also modify the platformio.ini file as follows (with debug_tool and debug_server - this should all be automatic):

[env:lpc1768]
platform = nxplpc
board = lpc1768
framework = mbed
debug_tool=custom
debug_server=pyocd-gdbserver

ivankravets, 2018-02-06:
Implemented in https://github.com/platformio/platform-nxplpc/commit/be0e0be4d97c6498c65a11ed89916e27e723f00d

Please re-test:

1. Open PIO IDE Terminal and type `pio upgrade --dev` and `pio update`
2. Remove all `debug_*` configuration items from `platformio.ini`.
3. Switch to a development version of NXP dev/platform: `platform = https://github.com/platformio/platform-nxplpc.git`

So, a default configuration for LPC1768, `upload_protocol = mbed`, it means that PIO will use mbed disk for firmware uploading. A default debugging tool is set to onboard `cmsis-dap` via `pyOCD`. No need to do extra steps. All should work our of the box.

Would be thankful if you try a new feature with a custom uploader. So, now you can instruct PIO to use pyOCD for flashing via `upload_protocol = cmsis-dap`.

P.S: @sstaub, you have this board. Please perform steps above and test both cases. Would be thankful for any feedback!

ivankravets, 2018-02-08:
Please help us. We want to release NXP LPC dev/platform. Could you re-test this with your LCP1768 board?

```
[env:lpc1768]
platform = https://github.com/platformio/platform-nxplpc.git
board = lpc1768
framework = mbed
upload_protocol = cmsis-dap
```

1. Does debugging work without extra configuration now?
2. Does uploading work via `cmsis-dap` directly?

Thanks!

wsnell, 2018-02-08:
Yes happy to. Will have feedback in the next two days.
-Wes

sstaub, 2018-02-09:
Hi Ivan,
upload via cmsis-dap works perfectly.
Starting the debug, i get following message:
[image]

with following errors:
[image]

when clicking on „Debug Anyway“ I can start the debugger and it works.

ivankravets, 2018-02-09:
@sstaub thanks a lot for the report! We will investigate issue with "Debug Anyway". 

> Also, when stoping the debugger and I want to restart the debug via statusbar
the cmd line pop ups instead of restarting directly the pro debugger.

See explanation here https://github.com/platformio/platformio-vscode-ide/issues/74#issuecomment-361704041

closed 2018-02-09 by ivankravets