platformio/platformio-vscode-ide #73: In debug mode local windo does not display float correctly

wsnell, 2018-01-13:
In debug mode local window does not display float correctly, but they do display correctly in the watch window.

See w=0 in local window and w=0.0049999999 in watch window.
See code below:
int main() {
    printf("This is a test\n");
    while(1) {
        float w=0.005;
        led2=(led2) ? 0 : 1;
        gpio1=1; gpio2=0; gpio3=0; gpio4=0; wait(w);
        gpio1=0; gpio2=1; gpio3=0; gpio4=0; wait(w+=.01);
        gpio1=0; gpio2=0; gpio3=1; gpio4=0; wait(w);
        gpio1=0; gpio2=0; gpio3=0; gpio4=1; wait(w);
        wait(.1);
    }
}

ivankravets, 2018-01-18:
Please report this issue to https://github.com/Microsoft/vscode-cpptools/issues

ivankravets, 2018-02-08:
Is this NXP dev/platform?

wsnell, 2018-02-08:
?Yes with mbed framework

ivankravets, 2018-04-07:
Should be fixed in https://github.com/WebFreak001/code-debug/commit/22198107157b32447cc73607f031f4ca1e34da89

Please navigate to VSCode extensions and check that "Native Debug" is up-to-date.

closed 2018-04-07 by ivankravets