platformio/platformio-vscode-ide #74: Should auto use the right debugger

wsnell, 2018-01-13:
When I load a platformio project and I cleck the "Debug/Start Debugging" it should use the coerrect debugger automatically.  It seems I have to set it each time.

Ideally the environment should be auto detected and ready to go right away, and all commands should just work.

ivankravets, 2018-01-18:
How many debug configurations do you have? By default, PIO generates 1 configuration which is set by default. It works for me without extra steps.

ivankravets, 2018-01-30:
Now I understand what do you mean. You click on "debug configuration" on the bottom toolbar. See https://code.visualstudio.com/docs/editor/debugging#_debug-in-status-bar

> a user can change the active launch configuration and then start debugging without the need to open the Debug view

This is a default behavior of this button in VSCode, we can't change it. I recommend to use GREEN arrow in debug mode.

[image]

closed 2018-01-30 by ivankravets

wsnell, 2018-01-30:
A couple: Windows, Python, Web, .NETCore. I find you have to do it every startup.

ivankravets, 2018-01-30:
Sorry, I don't understand you. Could you explain in details?