platformio/platformio-vscode-ide #76: Not working with custom shell

JensGuckenbiehl, 2018-01-17:
I use Cmder as integrated shell in Visual Studio Code with this settings:
```
    "terminal.integrated.shell.windows": "C:\\WINDOWS\\System32\\cmd.exe",
    "terminal.integrated.shellArgs.windows": ["/K", "C:\\Cmder\\vscode.bat"]
```

The vscode.bat:
```
@echo off
SET CurrentWorkingDirectory=%CD%
SET CMDER_ROOT=C:\Cmder
CALL "%CMDER_ROOT%\vendor\init.bat"
CD /D %CurrentWorkingDirectory%
```

When I use the PlatformIO Toolbar to build or open a terminal, "cmd" is used, but without the Cmder includes (example: no git)

ivankravets, 2018-05-26:
We don't remove/replace built-in VSCode terminal for the which you specified custom settings. Just use `Ctrl + `  ` hotkey or command palette and type for terminal.

JensGuckenbiehl, 2018-05-27:
PlatformIO extension (Build process, Upload process, open terminal via PlatformIO toolbar) use not the default terminal from VSCode. See "terminal.js" in line 23.
`return vscode.window.createTerminal({
      name: 'PlatformIO',
      shellPath: constants.IS_WINDOWS ? 'cmd.exe' : undefined,
      env: {
        PATH: process.env.PATH,
        PLATFORMIO_CALLER: 'vscode'
      }
});`

ivankravets, 2018-05-31:
We don't use SHELL for tasks, see https://github.com/platformio/platformio-vscode-ide/blob/develop/src/tasks.js#L238

commit e80ac105 (2018-07-10): Do not force PIO IDE Terminal to `cmd.exe` on Windows // Resolve #76
closed 2018-07-10 by ivankravets