platformio/platformio-vscode-ide #77: calling esptool.py runs on error when windows username contains a blank

cyberman54, 2018-01-21:
I'm using Windows 10 and my Windows account name contains a blank (let's say "Foo Bar").
Since a few days this seems to cause esptool.py to be called with wrong parameters:

```
Uploading .pioenvs\heltec_wifi_lora_32\firmware.bin
usage: esptool write_flash [-h] [--flash_freq {keep,40m,26m,20m,80m}]
[--flash_mode {keep,qio,qout,dio,dout}]
[--flash_size FLASH_SIZE]
[--spi-connection SPI_CONNECTION] [--no-progress]
[--verify] [--compress | --no-compress]
<address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Address "Foo Bar\Documents\PlatformIO\Projects\Paxcounter\.pioenvs\heltec_wifi_lora_32\partitions.bin" must be a number
*** [upload] Error 2
```

ivankravets, 2018-01-23:
http://docs.platformio.org/en/latest/platforms/espressif32.html#using-arduino-framework-with-staging-version

closed 2018-01-23 by ivankravets

cyberman54, 2018-01-28:
Thank you for your help. I changed platform to staging version, as described. Now i run into this error when trying to build my code:

`espressif32 @ 1392c2a has been successfully installed!
Uninstalling toolchain-xtensa32 @ 2.50200.0:    [OK]
The platform 'https://github.com/platformio/platform-espressif32.git#feature/stage' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
PackageManager: Installing toolchain-xtensa32 @ ~2.50200.0
Downloading  [------------------------------------]    0%
Error: Please read http://bit.ly/package-manager-ioerror
Warning! Package Mirror: [Errno 0] Error
Looking for another mirror...
Downloading  [------------------------------------]    0%
Error: Please read http://bit.ly/package-manager-ioerror
Warning! Package Mirror: [Errno 0] Error
Looking for another mirror...
Error: Could not install 'toolchain-xtensa32' with version requirements '~2.50200.0' for your system 'windows_amd64'.
More details: http://bit.ly/faq-package-manager`