platformio/platformio-vscode-ide #81: "Show on startup" checkbox not working on windows

simonszu, 2018-01-30:
I have a problem with PlatformIO IDE showing up in a foreground tab each time i start VS Code on Windows 10. Deactivating the "Show on startup" checkbox and restarting VS Code results in PlatformIO appearing again, with the checkbox re-activated. 
I do not have this problem on another host running VS Code on Linux.

ivankravets, 2018-01-30:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

simonszu, 2018-02-01:
```
H:\>echo %PATH%
C:\ProgramData\Oracle\Java\javapath;c:\Program Files (x86)\Intel\iCLS Client\;c:\Program Files\Intel\iCLS Client\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Windows\SysWOW64\Empirum;C:\Program Files (x86)\Enterprise Vault\EVClient\x64\;C:\Program Files\TortoiseGit\bin;C:\Program Files\Git\cmd;C:\Users\szustkos\AppData\Local\Programs\Python\Python36-32\Scripts\;C:\Users\szustkos\AppData\Local\Programs\Python\Python36-32\;C:\Users\szustkos\AppData\Local\Programs\Python\Python36\Scripts\;C:\Users\szustkos\AppData\Local\Programs\Python\Python36\;C:\Users\szustkos\AppData\Local\Microsoft\WindowsApps;C:\Users\szustkos\AppData\Local\Programs\MiKTeX 2.9\miktex\bin\x64\;C:\Program Files\Microsoft VS Code\bin

H:\>echo %PYTHONPATH%
%PYTHONPATH%

H:\>python --version
Python 3.6.3

H:\>pip --version
pip 9.0.1 from c:\users\szustkos\appdata\local\programs\python\python36-32\lib\site-packages (python 3.6)

H:\>pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
numpy (1.13.3)
pip (9.0.1)
scipy (1.0.0)
setuptools (28.8.0)

```

ivankravets, 2018-02-02:
Did you override `HOME` or `USERPROFILE` environment variables?

simonszu, 2018-02-02:
Not intentionally. I checked the environmental variables list in the windows system settings before answering to your questions. It is a work station connected to the company's AD, though, maybe it inherits some values from there. 
Can you tell me what these variables should look like? You seem to be surprised about their values.

ivankravets, 2018-02-02:
See

* https://github.com/platformio/platformio-node-helpers/blob/master/src/home.js#L144
* https://github.com/platformio/platformio-node-helpers/blob/master/src/core.js#L16

Where is located your `homestate.json`? Could you share its contents?

simonszu, 2018-02-05:
The `homestate.json`is located in `C:\Users\szustkos\.platformio` which is covered by `%USERPROFILE%\.platformio`.

These are its contents:
```
{"storage": {"homeDir": "C:\\Users\\szustkos", "coreCaller": "vscode", "cid": "658bf823-0720-e240-7e51-e86ac37d3d49", "projectsDir": "C:\\Users\\szustkos\\Documents\\PlatformIO\\Projects", "lastCheckPlatformUpdates": 1517820021671, "lastUpdateCorePackages": 1517820021670, "coreVersion": "3.5.2a7", "routeBadges": {}, "lastCheckLibraryUpdates": 1517820021670, "recentProjects": [], "coreSystype": "windows_amd64", "showOnStartup": {"vscode": false}}, "inputValues": {}}
```

I have removed Platformio during the weekend, restarted VScode and added Platformio again - it seems that this issue is resolved - but i'll keep an eye on it.

ivankravets, 2018-02-05:
Thanks a lot for your solution!

closed 2018-02-05 by ivankravets