platformio/platformio-vscode-ide #82: Support for cortex-m debug plugin

sstaub, 2018-01-30:
There is a new extension called Cortex-Debug [https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug](url) I think it would be nice to integrate it in PIO.

ivankravets, 2018-05-08:
Please upgrade to 0.15.0 https://github.com/platformio/platformio-vscode-ide/releases/tag/v0.15.0

Does it work for you?

closed 2018-05-08 by ivankravets

sstaub, 2018-05-08:
I tested it with a stm32 nucleo-f401re and it looks fine. Thank you, great work.
One thing I found is that tree view of a watched register collapsed after continue:
[image]
after "continue"
[image]

ivankravets, 2018-05-08:
Thank you so much for the feedback!

> One thing I found is that tree view of a watched register collapsed after continue:

Could you open separate issue?