platformio/platformio-vscode-ide #83: The MBED C/C++ framework library is old (150 but need 157+) needs to be updated

wsnell, 2018-01-30:
The MBED framework library is old and that has some significant bugs, such as the Ticker function do not work properly at all.  Run a couple of tickers and you will see the interrupts stall and then restart and can lock up the system for 500msecs and see random timer events.

Running my same code on the MBED online compiler does not have this problem, but compile and run under the very latest PlatformIO tools (Jan 18 2018 or so), it fails (see above).

I tracked this down to the fact that the MBED frame is using an old 150 release whereas the latest MBED provided library is now 157+.   So can you guys simply replace the old 150 library with 157? It is critical and will cause us not to use the IDE until fixed (although if you can provide a workaround how to replace the library locally I can try that too).

In short would like you to update the MBED lib to 157+.  Should be as simple as replacing the files provided in the platform package....
See the MBED versions here: https://os.mbed.com/users/mbed_official/code/mbed/

ivankravets, 2018-01-30:
What is your dev/platform? We use the latest mbed. I know that Atmel SAM is fixed to very old mbed because Microchip left ARM mbed.

Please provide your `platformio.ini`.

ivankravets, 2018-02-08:
> But why are you guys still shipping via the normal setup a very old library (non GIT)

We don't ship very old mbed OS. We ship the latest which were tested by our team and community. If we close all issues, then we make a final release.

Anytime you can use the latest dev/platform and related framework using `platform = https://github.com/platformio/platform-nxplpc.git`.

The final release is coming soon.

> By the way otherwise love this product.

Thank you very much! 

P.S: Would be thankful if update your rating [here](https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide#review-details). We are working on the fixing all issues which you reported.

wsnell, 2018-02-08:
?One more comment.  I never knew about the dev branch and after very much digging I found this reference.  It would be nice if you can make that info more prevalent, especially in the platform.ini file docs (with examples).

Thx

wsnell, 2018-02-08:
My review...

This is an outstanding product.  It still has a few quirks but they are working quickly on them. Also they have support for a massive amount of platforms too.
Hey guys can you also make it work with Visual Studio 2017?
?

ivankravets, 2018-02-09:
I hoped you can edit 😊 
[image]

ivankravets, 2018-02-09:
>  It would be nice if you can make that info more prevalent, especially in the platform.ini file docs (with examples).

We have that:

* http://docs.platformio.org/en/latest/projectconf/section_env_general.html#platform
* http://docs.platformio.org/en/latest/platforms/nxplpc.html#stable-and-upstream-versions

In any case, I close this issue, we will release NXPLPC soon.

closed 2018-02-09 by ivankravets