platformio/platformio-vscode-ide #84: ESP8266 Unable to upload in VSCODE, ATOM IDE works.

leodesigner, 2018-01-31:
Hi,
First of all thank you for such an amazing project (Platformio).

Due some issues in uploader I am not able to upload any compiled binary to NodeMCU board from VSCODE IDE, 

(Home 0.6.1·Core 3.5.2a7)

```
...
Available:
Looking for upload port...
Auto-detected: /dev/cu.wchusbserial1410
Uploading .pioenvs/esp01/firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error 255
========================================================== [ERROR] Took 5.05 seconds ==========================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

```

However from ATOM IDE everything works.
(Using MacOS El Capitan)

Дякую.

ivankravets, 2018-01-31:
Please re-install USB driver for your USB/UART chip. Also, try to use another cable or switch to another USB port.

https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

> Дякую.

Будь ласка 😊

leodesigner, 2018-01-31:
I am using NodeMcu with ch340g serial chip.
I have AtomIDE and VSCODE IDE installed on the same machine at the same time. 
So they are using the same ch340g driver. Also I was using the ch340g serial adapter with ESP8266-01.
It is really strange, because I think AtomIDE and VSCODE both are calling "pio run --target upload" under the hood. It seems like I was able to upload once when I was resetting ESP board in the right time just before start of the upload procedure.

ivankravets, 2018-01-31:
Ah...stop.. It works with Atom...

Please open PIO IDE Terminal in VSCode (terminal icon on the left bottom toolbar) and type
```
pio run -t upload
```

Does it work?

leodesigner, 2018-02-01:
It's funny but I cannot reproduce this bug now. 
(after applying latest security patches for MacOS)
Now everything is working, both from terminal and from VSCODE IDE.
I will test this more with serial adapter and ESP-01 module later today.
For now I think I will close the issue, sorry for bothering :)

closed 2018-02-01 by leodesigner