platformio/platformio-vscode-ide #85: Old ARM GDB has a bug setting assembly breakpoint

johnjiabinzhang, 2018-02-08:
Hi, I'm just wondering if PlatformIO IDE can update the ARM GCC tool-chain version that it's downloading. The current version
```
GNU gdb (GNU Tools for ARM Embedded Processors) 7.6.0.20140731-cvs
Copyright (C) 2013 Free Software Foundation, Inc.
```
is very old. In our use case there seems to be a bug in GDB that when I try to set a break point at the beginning of an assembly function, it fails to set the break point at the correct instruction address.

Though the same issue is not fixed in the most recent version,
```
GNU gdb (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 8.0.50.20171128-git
Copyright (C) 2017 Free Software Foundation, Inc.
```
there is a walk-around in setting a function break-point. However this walk-around doesn't work in the old version.

Also the newer version comes with a Python scripting enabled GDB, which allows for support for more debug varieties.

So would you guys be able to update the ARM GCC toolchain version? Thanks!

ivankravets, 2018-02-08:
Which platform/board do you use?

johnjiabinzhang, 2018-02-08:
STM32L476vg Discovery board on macOS.

johnjiabinzhang, 2018-02-08:
I wrote an issue on our own issue tracker. I can't paste the link here because it's not publically accessible. But I will re-post my findings here:

## Description
This is a GDB issue, and it seems to only affect setting break points at **beginning of a function**.

When attempting to set a break point at the start of the function (line 65), in debug mode it actually breaks at the next instruction (line 68):
[image]

There is a problem in how GDB sets the break point, explained below.

The in the disassembly of the ELF file, we can see that the start of main function is at address `0x080001dc`:
```
080001dc <main>:
$t():
/Users/johnz/CourseFiles/COMP2300/pio-stm32cube-template/src/main.S:65
 80001dc:       492b            ldr     r1, [pc, #172]  ; (800028c <countdown10+0xc>)
/Users/johnz/CourseFiles/COMP2300/pio-stm32cube-template/src/main.S:68
 80001de:       6ccb            ldr     r3, [r1, #76]   ; 0x4c
 80001e0:       f04f 0001       mov.w   r0, #1
 80001e4:       ea4f 0040       mov.w   r0, r0, lsl #1
 80001e8:       ea43 0300       orr.w   r3, r3, r0
 80001ec:       64cb            str     r3, [r1, #76]   ; 0x4c
```

If I manually add break point through MI commands it gives the following reply:
```
> -break-insert main.S:65
{...["addr","0x080001de"],["file","src/main.S"],...,["line","65"],...,["original-location","main.S:65"]]]]}}
```
This says that a break point is inserted at address `0x080001de`, which is the next instruction, rather than `0x080001dc`.

I believe this is related to a [bug](https://sourceware.org/bugzilla/show_bug.cgi?id=17036) that has been reported a few years back and it still hasn't been resolved.

## Solution
There is a walk around to this problem by setting a **function break point** rather than a **line break point**.

In VSCode, a function break point can be add by clicking the plus button on the BREAK POINTS view:
[image]

Then type in the function name, and it should break at the beginning of the function.
This is verified through the MI interface:
```
> -break-insert main
{...["addr","0x080001dc"],["file","src/main.S"],...,["line","65"],...}
```

## Caveat
__This seems to work only in a *more recent version of ARM GDB*!__

The ARM GCC toolchain that PlatformIO downloads is too old:
```
GNU gdb (GNU Tools for ARM Embedded Processors) 7.6.0.20140731-cvs
Copyright (C) 2013 Free Software Foundation, Inc.
```
It's a 2014 July build. The most recent version is 2017 November build:
```
GNU gdb (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 8.0.50.20171128-git
Copyright (C) 2017 Free Software Foundation, Inc.
```

The above walk-around **does not** work on the old version, but works on the new version. But unfortunately the new version still hasn't fixed the issue, but somehow the walk-around works.

The newer ARM GCC can be installed on macOS by doing
```
$ brew cask install gcc-arm-embedded
```
Can it can be found [here](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads) on Windows.

wsnell, 2018-02-08:
NXP ?LPC1768 on mbed

ivankravets, 2018-02-14:
Thanks! All toolchains were updated. Please open `PIO Home > Platforms > Updates`

closed 2018-02-14 by ivankravets

johnjiabinzhang, 2018-02-14:
Great thanks!

johnjiabinzhang, 2018-02-15:
macOS.

johnjiabinzhang, 2018-02-15:
```bash
$ ls -al ~/.platformio/packages
total 0
drwxr-xr-x  11 johnz  staff   352 15 Feb 22:19 .
drwxr-xr-x  11 johnz  staff   352 15 Feb 22:19 ..
drwx------   9 johnz  staff   288 11 Jan 09:30 contrib-piohome
drwx------  62 johnz  staff  1984 11 Jan 09:31 contrib-pysite
drwx------  20 johnz  staff   640  1 Dec 09:58 pysite-pioplus
drwx------   4 johnz  staff   128 14 Feb 14:08 tool-pioplus
drwx------  16 johnz  staff   512  1 Feb 12:06 tool-scons
drwx------   9 johnz  staff   288 15 Feb 22:19 tool-stlink
drwx------  12 johnz  staff   384 15 Feb 22:19 tool-stm32duino
drwx------   8 johnz  staff   256 11 Jan 11:20 tool-unity
drwx------   8 johnz  staff   256 15 Feb 22:18 toolchain-gccarmnoneeabi
```

ivankravets, 2018-02-15:
`cat ~/.platformio/packages/toolchain-gccarmnoneeabi/package.json`

johnjiabinzhang, 2018-02-15:
```
$ cat ~/.platformio/packages/toolchain-gccarmnoneeabi/package.json
{
    "description": "gcc-arm-embedded",
    "name": "toolchain-gccarmnoneeabi",
    "system": [
        "darwin_x86_64",
        "darwin_i386"
    ],
    "url": "https://launchpad.net/gcc-arm-embedded",
    "version": "1.40804.0"
}
```

ivankravets, 2018-02-15:
I got you... Please provide `platformio.ini`. New toolchain will be available only for mbed framework. We don't test it with Arduino libopencm3 frameworks. I know that Arduino Core for STM32 uses old toolchain.

johnjiabinzhang, 2018-02-15:
```
[env:disco_l476vg]
platform = ststm32
board = disco_l476vg
build_flags = -Wl,-T"lib/bare_stm32l476/ldscripts/STM32L476VG_FLASH.ld"
build_unflags = -nostartfiles -nostdlib
```

ivankravets, 2018-02-15:
framework?

johnjiabinzhang, 2018-02-15:
For teaching purpose at this moment we are doing a very minimum bare build. But in our assignment repo we are using `stm32cube`.
```
[env:disco_l476vg]
platform = ststm32
board = disco_l476vg
framework = stm32cube
```

ivankravets, 2018-02-15:
I've just enabled GCC 7 toolchain for all frameworks in ST STM32 dev/platform. Please try upstream version http://docs.platformio.org/en/latest/platforms/ststm32.html#stable-and-upstream-versions

johnjiabinzhang, 2018-02-16:
Great, thanks!
So this update will be in the next release, right? Approximately when would that be?

johnjiabinzhang, 2018-02-18:
We will be using PlatformIO for a university course. We are starting in this coming week. It would be great if there can be a new release soon.

ivankravets, 2018-02-19:
We have done a few major updates and are so close to the final release. Could you re-test the latest upstream version? 

http://docs.platformio.org/en/latest/platforms/ststm32.html#stable-and-upstream-versions